arduino/Arduino #6438: Feature request: Line Highlight

aliirmak, 2017-06-29:
Can you add an option for highlighting current line? It helps some of us where we are typing.

per1234, 2017-06-29:
Maybe you're asking if it can be added as a preference but this is already possible by editing the theme file:
1. Open the file lib/theme/theme.txt found in the Arduino IDE installation folder.
2. Change the line:
```
editor.linehighlight=false
```
to:
```
editor.linehighlight=true
```
3. Save the file.
4. Restart the Arduino IDE if it's running.

You can also change the color of the highlight via `editor.linehighlight.color`

facchinm, 2017-07-03:
Thanks @per1234 for the super precise comment! I'm leaving the issue open to understand if the request was indeed to add a checkbox in Preferences or if writing it in some wiki page is enough.

aliirmak, 2017-07-03:
Yes. You both are pnting to the right direction.
A checkbox under Preferences next to Line Numbering would be nice.