arduino/Arduino #6439: Arduino UNO win 10 WILL not find the board.

mortenwallinjensen, 2017-06-29:
Is ther somewhere I can get the propper USB/com driver for my Arduino UNO board? Windows 10 have finally owerwritten / owerrouled the Arduino driver with one it tells me it can not use. Failure 43...

matthijskooijman, 2017-06-29:
Windows 10 should work without installing any drivers (it should automatically select the generic CDC ACM driver). Are you using an official Arduino, or a third party clone or counterfeit?

mortenwallinjensen, 2017-06-29:
I'm using an official Arduino UNO. Yesterday I could communicate with it on COM5. But that was after I han tried 3 different USB slots. But it stopped working after I pulled out the USB cable and later reinsertet it.

facchinm, 2017-07-03:
Hi @mortenwallinjensen ,
if the board was working before it's very unlikely that later became faulty... It's probably related with a problem on the PC USB side. Try changing USB port and eventually, if itstill doesn't work, feel free to contact the [support](support@arduino.cc), thanks!

closed 2017-07-03 by facchinm