arduino/Arduino #6444: Compilation fails (generating function prototypes?)

bobsuname, 2017-07-01:
Here's the last part of the verbose output (same problem occurs if compiling for UNO):
.....
Generating function prototypes...
"C:\Users\Bob Gilman\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_PROMICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x1b4f -DUSB_PID=0x9206 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="SparkFun Pro Micro"' "-IC:\Users\Bob Gilman\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Users\Bob Gilman\AppData\Local\Arduino15\packages\SparkFun\hardware\avr\1.1.5\variants\promicro" "C:\Users\BOBGIL~1\AppData\Local\Temp\arduino_build_399105\sketch\TestEDEKeypad.ino.cpp" -o "C:\Users\BOBGIL~1\AppData\Local\Temp\arduino_build_399105\preproc\ctags_target_for_gcc_minus_e.cpp"
exit status -1073741515
Error compiling for board SparkFun Pro Micro.
.........
Not much of a clue.  I found a bad prototype in a library header file that isn't used by the sketch I was trying to compile, but fixing it didn't change the failure.  Ideas?

matthijskooijman, 2017-07-01:
Could you paste a full (but preferably reduced as much as possible) sketch that shows the problem, including any libraries needed?

bobsuname, 2017-07-01:
I suspect the problem is not in the particular libraries used (unless it's the Serial library).   I get the same error when I try to compile the following:
---------------
// test operations with ints and unsigned longs

unsigned long time, result;
int delta;

void setup()
{
  time = 17000000UL;
  delta = 222;
  result = 0;
  
  Serial.begin(9600);
  while (!Serial);
  Serial.println("Hit return to start");
  while(Serial.read() < 0);
  result = time + delta;
  printit(" plus ");
  result = time - delta;
  printit(" minus ");
  delta = -delta;
  result = time + delta;
  printit(" plus ");
  result = time - delta;
  printit(" minus ");
}

void printit(char *str)
{

  Serial.print(result);
  Serial.print(" = ");
  Serial.print(time);
  Serial.print(str);
  Serial.println(delta);
  return;
}

void loop()
{
  ; // do nothing
}
--------------
Here's the entire verbose output from the compilation
-------------
Arduino: 1.8.2 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Bob Gilman\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Bob Gilman\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Bob Gilman\Documents\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0001 -ide-version=10802 -build-path C:\Users\BOBGIL~1\AppData\Local\Temp\arduino_build_389000 -warnings=default -build-cache C:\Users\BOBGIL~1\AppData\Local\Temp\arduino_cache_590124 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\Bob Gilman\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Bob Gilman\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avrdude.path=C:\Users\Bob Gilman\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -verbose C:\Users\Bob Gilman\Documents\Arduino\sketches\ulongtest\ulongtest.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Bob Gilman\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Bob Gilman\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Bob Gilman\Documents\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0001 -ide-version=10802 -build-path C:\Users\BOBGIL~1\AppData\Local\Temp\arduino_build_389000 -warnings=default -build-cache C:\Users\BOBGIL~1\AppData\Local\Temp\arduino_cache_590124 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\Bob Gilman\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Bob Gilman\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avrdude.path=C:\Users\Bob Gilman\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -verbose C:\Users\Bob Gilman\Documents\Arduino\sketches\ulongtest\ulongtest.ino
Using board 'uno' from platform in folder: C:\Users\Bob Gilman\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19
Using core 'arduino' from platform in folder: C:\Users\Bob Gilman\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19
Detecting libraries used...
"C:\Users\Bob Gilman\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Bob Gilman\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Users\Bob Gilman\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\variants\standard" "C:\Users\BOBGIL~1\AppData\Local\Temp\arduino_build_389000\sketch\ulongtest.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\Bob Gilman\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Bob Gilman\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Users\Bob Gilman\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\variants\standard" "C:\Users\BOBGIL~1\AppData\Local\Temp\arduino_build_389000\sketch\ulongtest.ino.cpp" -o "C:\Users\BOBGIL~1\AppData\Local\Temp\arduino_build_389000\preproc\ctags_target_for_gcc_minus_e.cpp"
exit status -1073741515
Error compiling for board Arduino/Genuino Uno.
--------------------------

bobsuname, 2017-07-02:
It gets better: the following sketch fails in the same way (same error code):
----------
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
I tried reinstalling the IDE (both 1.8.2 and 1.8.3) to no avail.

bobsuname, 2017-07-04:
I tried recreating the ../AppData/Local/Arduino15 directory, and this seems to have worked.  i'm not sure what got messed up therein.

closed 2017-07-04 by bobsuname