arduino/Arduino #6450: [new library] Please add "aws-sdk-arduino-esp8266"

Schm1tz1, 2017-07-02:
I have forked and cleaned-up some of the AWS-IoT-Code found here on GitHub and brought it to 1.5-format.
https://github.com/Schm1tz1/aws-sdk-arduino-esp8266

per1234, 2017-07-02:
You need to create a release or tag that matches the `version` value in your library.properties file.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Schm1tz1, 2017-07-02:
Sorry - created the release after the file. Done, the two versions match now. Thanks a lot !

per1234, 2017-07-02:
Your release v0.9-beta still has the incorrect `version` value in library.properties because the commit that changes that value was pushed after the release:
https://github.com/Schm1tz1/aws-sdk-arduino-esp8266/blob/v0.9-beta/library.properties#L2

The release must contain the commit that changes the `version` value to match the release version. You will need to either create a new release or redo the v0.9-beta release.

Schm1tz1, 2017-07-02:
Ok, tagged a new release now. Sorry for the inconvenience.
https://github.com/Schm1tz1/aws-sdk-arduino-esp8266/blob/0.9.1-beta/library.properties

cmaglie, 2017-07-06:
Done!

closed 2017-07-06 by cmaglie