arduino/Arduino #6454: Make it easier to edit the Arduino IDE theme

per1234, 2017-07-03:
This topic has come up previously in https://github.com/arduino/Arduino/issues/3056, https://github.com/arduino/Arduino/issues/6025 and https://github.com/arduino/Arduino/issues/6438 but the latter two topics are more focused on controlling specific properties via the GUI.

This also comes up regularly on the Arduino forum and, as with https://github.com/arduino/Arduino/issues/3056 and https://github.com/arduino/Arduino/issues/6025, it's almost always an accessibility issue rather than aesthetics so this is important.

I'm sure most users would prefer some sort of a GUI theme editor but I don't know how feasible that would be. This could be a standalone application or added to **File > Preferences**. Offering even the most minimal control of the theme would end up adding quite a few items to Preferences so it would probably be best to add a separate tab, as is already done with the Network settings.

A more simple, if less user friendly option, is to just document the location of the theme files and the function of the various properties. I have made a start at this in the readme of https://github.com/per1234/ThemeTest but it needs some polishing to make it more accessible to the average user. Since this is more advanced information, it might make sense to add it as a page on the [GitHub wiki](https://github.com/arduino/Arduino/wiki). The Arduino Playground is another option.

A significant problem is that the theme files are stored in the Arduino IDE installation folder and thus any changes to those files will be lost after upgrading to a new version. The solution is to move these files to a persistent location such as the Arduino15 folder, as is already done with preferences.txt.

pacav69, 2018-10-05:
Here is a workaround.
orignally posted here https://github.com/arduino/Arduino/issues/7451#issuecomment-427275339
@jockm i totally agree with you that the color for errors of orange/red whatever should be a more distinct color for those with difficulty in reading.
So i have put together some information on changing some parameters in theme.txt that will make the text in the IDE more easily read.

[image]

[changing_error_color.txt](https://github.com/arduino/Arduino/files/2449326/changing_error_color.txt)

animeshsrivastava24, 2019-05-28:
I am looking forward to work for enhancing and improving the theme work, and adding relevant information to the README.md on https://github.com/per1234/ThemeTest/blob/master/README.md . Please provide me further suggestions @per1234

per1234, 2019-05-28:
>and adding relevant information to the README.md on https://github.com/per1234/ThemeTest/blob/master/README.md . Please provide me further suggestions @per1234

Great! Contributions to my ThemeTest repository are welcome. You can submit pull requests to that repository for any improvements or corrections you would like to propose.

per1234, 2019-05-29:
Responding to https://github.com/arduino/Arduino/pull/7115#issuecomment-496923890

>I would like to work on following topics and produce the result as a PR.
>1. Designing a theme editor for the Arduino as mentioned in the Option 1 above

That would be wonderful!

>Documenting each part of the text in the file .arduino/preference.txt

I agree that it would make sense to also document any relevant parts of preferences.txt (e.g., `editor.font`) in the Arduino IDE themes documentation.

>Creating a user-guide for the same that can be published as an article- the features of the user-guide will be more like a book with chapters moving from top to bottom of the GUI of the arduino and indicating the methods to change each part and features in detail with help of Proper Color Format and patterns

That would be very useful. We don't have any way to submit this sort of documentation as a PR. My idea is that it would make sense to add a page documenting Arduino IDE themes to this repository's wiki, since it is documentation for theme developers. My suggestion would be for you to create your own GitHub repository and first publish the documentation you write in the wiki of that repository. The initial development and collaboration on the content can occur there. Once the documentation is in a finished state, you can present a link to your wiki for Arduino's approval, after which you can copy the content over to Arduino's wiki.

>Should I open it as an Issue(Work) with the Name: "GUI Theme Editor and Documentation".

We already have this issue (https://github.com/arduino/Arduino/issues/6454) on that topic so there is no need to create another topic.

---
Note that if the work you are proposing is intended to be part of [Arduino's Season of Docs program](https://github.com/arduino/season-of-docs), you must still go through the [formal process established by Google](https://developers.google.com/season-of-docs/docs/tech-writer-guide) before starting the GSoD project work. Now is an appropriate time for you to be doing the research that will lead to the project proposal in your application, but the actual "Doc development" phase of the program does not start until 2019-11-02. If you just want to do the work as a volunteer outside of the Season of Docs program, that is very welcome and you can start whenever you like.

animeshsrivastava24, 2019-05-29:
@per1234  I have designed a basic Wiki Page for the Theme, from my research. Please have a look- I'll keep on updating the content to add the most significant and important points:
https://github.com/animeshsrivastava24/ArduinoIDE-Theme/wiki