arduino/Arduino #6462: [Library Manager]: Adafruit Circuit Playground Library

Gohstgamer, 2017-07-05:
Hi there,

After a certain time after starting up the Arduino Application, there comes this little "Banner" on the bottom of the program which indicates there are some librarys not up to date. If I click on that I see Adafruit Circuit Playground Version 1.6.4, after updating and reopening the Library Manager it is still there, I also reinstalled allready the whole Arduino IDE (nothing changed). But in the /Arduino/libraries/ there appears everytime after "updating" a new folder with a Name like this "arduino_xxxxxx" (each x stays for one number between 0-9).

Further Information:
Arduino 1.8.3 IDE
Mac OS 10.12.5 on a MBP Retina 15" (Mid 2014)

Greets Simon

per1234, 2017-07-05:
What do you have selected in the **Tools > Board** menu?

Gohstgamer, 2017-07-05:
Arduino/GenuinoMega 2560 (ATmega 2560) and Arduino/Genuino UNO

per1234, 2017-07-05:
I can reproduce this.

The cause is that Adafruit has changed the library `name` value in library.properties from "Adafruit Circuit Playground" to "Adafruit CircuitPlayground" since the 1.6.4 version bundled with Arduino IDE 1.8.3:
https://github.com/adafruit/Adafruit_CircuitPlayground/blob/1.6.4/library.properties#L1
https://github.com/adafruit/Adafruit_CircuitPlayground/blob/1.6.7/library.properties#L1

ladyada, 2017-07-09:
yah this is actually in an attempt to fix a problem we found where you can have the built-in/default "Adafruit_CircuitPlayground" and then when you try to update it, the new library is called "Adafruit_Circuit_Playground" and the compiler keeps using the old version. Since we couldn't figure out where the extra _ came from, we tried to rename the library.properties

@cmaglie do you have any suggestions/way on how we can make sure that
https://github.com/adafruit/Adafruit_CircuitPlayground/blob/master/library.properties
is always installed as the folder "Adafruit_CircuitPlayground"? that will allow people to update the built-in version and have the new folder properly named so that it comes first in compilation :/

bill-orange, 2017-07-11:
Currently, is there a fix for this?  I get the update banner most every time I start the IDE.

per1234, 2017-07-11:
Here's the workaround:

- File > Examples > Adafruit Circuit Playground > accel_mouse
- Sketch > Show sketch folder
- Navigate up two folder levels
- Open the file library.properties in a text editor.
- Change the line:

```
name=Adafruit CircuitPlayground
```
to:
```
name=Adafruit Circuit Playground
```
- Save the file.

Alternative workaround:
- **File > Preferences > Check for updates on startup (uncheck) > OK** - be aware that this turns off update notifications for all libraries and Boards Manager hardware packages (and IDE update notifications too if that ever gets fixed).

Neither will solve the problem ladyada mentioned of the Arduino IDE using the previous library version due to the header filename/folder name mismatch of the library version installed via Library Manager. It's only a quick fix for the always updatable annoyance. Hopefully the root problem will be solved in time.

ladyada, 2017-07-17:
hiya @per1234 @cmaglie - who can tweak the Circuit Playground Library Manager entry so that it downloads into "Adafruit_CircuitPlayground"? we dont mind changing the name/properties as long as we can come up with a consistent setup.
thx!

cmaglie, 2017-07-18:
> yah this is actually in an attempt to fix a problem we found where you can have the built-in/default "Adafruit_CircuitPlayground" and then when you try to update it, the new library is called [...]

The extra `_` comes from the `name` field in library.properties: `Adafruit Circuit Playground` is converted to `Adafruit_Circuit_Playground` (spaces are turned into underscores) and that's the correct folder name that should be used, I would not touch that.

The fact that the compiler keeps using the old version is really weird, so I checked more deeply what's going on. I've found that:

1) the library bundled in the IDE has the **folder** named `Adafruit_CircuitPlayground` but the `name` field in library.properties is `Adafruit Circuit Playground`, see 43bec97da7. I don't know if the name has been changed in the meantime or I've written it wrong in the first place, it could really be that I've mistyped being confused by the name of the repository.
2) the main include of the library is `Adafruit_CircuitPlayground.h`.
3) when the user upgrade the library through the library manager, the (new) library is unpacked in the sketchbook libraries folder inside a folder called `Adafruit_Circuit_Playground`.

it's the combination of 1, 2 and 3 that create the chaos here, becase when a sketch tries to include

`#include <Adafruit_CircuitPlayground.h>`

the compiler gives priority to the (old) library that is bundled with the IDE since it has the folder name that _match exactly_ with the include: `Adafruit_CircuitPlayground`.

Unfortunately changing the name of the library inside the library manager won't fix the problem because the IDE will recognize `Adafruit CircuitPlayground` as a different library from `Adafruit Circuit Playground` and the user won't be notified of the upgrade (and also creates the messy situation where two different libraries are displayed in the library manager).

This is a very bad situation, so here some possible ways out that comes to my mind:

**SOLUTION 1**: Rename `Adafruit_CircuitPlayground.h` to `Adafruit_Circut_Playground.h` and add a placeholder `Adafruit_CircuitPlayground.h` that just includes `Adafruit_Circut_Playground.h`. All the examples should be changed to use `#include <Adafruit_Circuit_Playground.h>`.

This may be not optimal, but it's doable immediately.

**SOLUTION 2**: Fix the bundled library in the IDE to have the correct folder name `Adafruit_Circuit_Playground`.

This will resolve the problem once for all, but needs a release of the IDE and users to upgrade their installation (as a side note: this is the curse of the bundled libraries...).

One solution doesn't exclude the other, I can apply immediately the SOLUTION 2, but as I said it will require an IDE release (that should happen in a couple of days BTW).

ladyada, 2017-07-18:
i think the confusion came because the name of the library is not the same as the name in the properties, my fault - i never realized this had to match up.
ill update the header and cpp file right now on the library bundle and push and update. we'll find out in an hour if it worked :D

ladyada, 2017-07-18:
ok ^ made a release. will try in an hour to see if it fixes things :) thank you for your help!!!

ladyada, 2017-07-18:
@cmaglie ok i had to fix a few other bugs so i pushed a minor release to the library. i verified that new code that #include "Adafruit_Circuit_Playground.h" works, all of the older  #include "Adafruit_CircuitPlayground.h" code still doesnt. Theres dozens of examples/tutorials, so i think i'd like to just hold out for Solution #2 if you dont mind! :)

commit e1594822 (2017-07-19): Fixed wrong folder name for Adafruit Circuit Playground library

Fix #6462
closed 2017-07-19 by cmaglie