arduino/Arduino #6463: Uploading via CLI lacks runtime.platform.path

Sorunome, 2017-07-06:
When uploading via CLI it lacks `runtime.platform.path`

e.g.

```
$ arduino GraphicTest/GraphicTest.ino -v --upload --useprogrammer
<compiling output>
Uploading...
/home/sorunome/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino6-static/bin/openocd -d2 -s /home/sorunome/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino6-static/share/openocd/scripts/ -f {runtime.platform.path}/variants/arduino_zero/openocd_scripts/arduino_zero.cfg -c telnet_port disabled; program {{/tmp/arduino_build_20957/GraphicTest.ino.elf}} verify reset; shutdown 
An error occurred while uploading the sketch
Open On-Chip Debugger 0.9.0-g932ec70 (2017-02-15-11:32)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Runtime Error: embedded:startup.tcl:60: Can't find {runtime.platform.path}/variants/arduino_zero/openocd_scripts/arduino_zero.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 60
```

running `arduino --get-prefs` doesn't show it, either

cmaglie, 2017-07-14:
You must specify a board with the `--board` option. The `runtime.platform.path` depends on the core used that is known only after a board is selected.
For example:

```
./arduino --board arduino:avr:uno  --get-pref runtime.platform.path
```

on my machine gives:

```
/home/cmaglie/Code/arduino/build/linux/work/hardware/arduino/avr
```

that is where the AVR core is installed.

Sorunome, 2017-07-14:
Well, when i don't specify --board it uses whatever i set in the gui so i'd expect to set runtime.platform.path accordingly

cmaglie, 2017-07-14:
Can you paste the command line you use to upload/verify without the `--board` option?

Sorunome, 2017-07-14:
It's in top post, `arduino GraphicTest/GraphicTest.ino -v --upload --useprogrammer`

PR #6529 (2017-07-27): CLI: fixed missing "runtime.platform.path"
This happens when the CLI is runned without the `--board` option.

Fix #6463 

/cc @Sorunome
commit 93217c40 (2017-07-27): CLI: fixed missing "runtime.platform.path"

This happens when the CLI is runned without the `--board` option.

Fix #6463
closed 2017-07-27 by cmaglie