arduino/Arduino #6466: ATmega328P is incorrectly called "ATmega328" in multiple places on arduino.cc

per1234, 2017-07-06:
As stated in https://github.com/arduino/Arduino/issues/6369, this causes much confusion, suffering and generally makes Arduino look bad. Let's use the correct name! That extra "P" is really not so scary is it?

The following pages incorrectly call the ATmega328P "ATmega328"
- https://cdn.arduino.cc/reference/en/language/functions/analog-io/analogreference/
- https://store.arduino.cc/usa/arduino-ethernet-rev3-with-poe (overview, tech specs, documentation tabs)
- https://store.arduino.cc/usa/arduino-ethernet-rev3-without-poe
- https://store.arduino.cc/usa/arduino-mini-05 (overview, tech specs, documentation tabs)
- https://store.arduino.cc/usa/arduino-nano (overview tab, tech specs tab, documentation tab)
- https://store.arduino.cc/usa/arduino-uno-rev3 (documentation tab)
- https://store.arduino.cc/usa/arduino-uno-rev3-with-long-pins
- https://store.arduino.cc/usa/arduino/arduino-boards-modules/arduino-ethernet-rev3-with-poe (overview, tech specs, documentation tabs)
- https://store.arduino.cc/usa/arduino/arduino-ethernet-rev3-with-poe  (overview, tech specs, documentation tabs)
- https://www.arduino.cc/en/Guide/ArduinoBT
- https://www.arduino.cc/en/Guide/ArduinoMini
- https://www.arduino.cc/en/Guide/ArduinoNano
- https://www.arduino.cc/en/Guide/ArduinoPro
- https://www.arduino.cc/en/Guide/ArduinoProMini
- https://www.arduino.cc/en/Guide/Environment
- https://www.arduino.cc/en/Guide/Environment0018
- https://www.arduino.cc/en/Guide/HowtoBoard
- https://www.arduino.cc/en/Guide/LilyPadMacOSX
- https://www.arduino.cc/en/Guide/LilyPadWindows
- https://www.arduino.cc/en/Guide/Troubleshooting
- https://www.arduino.cc/en/Hacking/HomePage
- https://www.arduino.cc/en/Hacking/PinMapping
- https://www.arduino.cc/en/Hacking/PinMapping168
- https://www.arduino.cc/en/Main/ArduinoBoardBT?from=Main.ArduinoBoardBluetooth
- https://www.arduino.cc/en/Main/ArduinoBoardDuemilanove
- https://www.arduino.cc/en/Main/ArduinoBoardFioProgramming
- https://www.arduino.cc/en/Main/ArduinoBoardFioProgramming2
- https://www.arduino.cc/en/Main/ArduinoBoardLilyPad (says "ATmega328V" but if you look at the product photo sure enough it's a ATmega328P!)
- https://www.arduino.cc/en/Main/ArduinoBoardLilyPadSimple
- https://www.arduino.cc/en/Main/ArduinoBoardProMini
- https://www.arduino.cc/en/Main/ArduinoBoardUnoSMD
- https://www.arduino.cc/en/Main/Boards (Arduino LilyPad says "ATmega328V", Arduino Nano)
- https://www.arduino.cc/en/Main/Standalone
- https://www.arduino.cc/en/Reference/AnalogReference
- https://www.arduino.cc/en/Reference/AnalogWrite
- https://www.arduino.cc/en/Reference/Board
- https://www.arduino.cc/en/Reference/EEPROM
- https://www.arduino.cc/en/Tutorial/AnalogInputPins
- https://www.arduino.cc/en/Tutorial/ArduinoToBreadboard
- https://www.arduino.cc/en/Tutorial/Memory
- https://www.arduino.cc/en/Tutorial/SecretsOfArduinoPWM
- https://www.arduino.cc/en/uploads/Main/Arduino-Pro-Mini-schematic.pdf
- https://www.arduino.cc/en/uploads/Main/ArduinoNano30Schematic.pdf

kengdahl, 2017-12-27:
Fixed most of the mentions of the 328 without the P. The rest are not that important. Closing the Issue.

per1234, 2018-08-01:
I strongly disagree. If Arduino can't even get the model number of the microcontroller right, what kind of quality can the customer expect from their products?