arduino/Arduino #6472: High DPI unusable on Linux

SvenDowideit, 2017-07-08:
Can't select the menu items, or edit with starting the v1.8.3 IDE on a HDPI screen on Linux (GNOME)

It looks like #4376 added a preference that we don't see on Linux?

PaulStoffregen, 2017-07-08:
Click File > Preferences.  Or hold the CTRL key and press the comma ',' key.

[image]

PaulStoffregen, 2017-07-08:
These definitely do work on my Ubuntu system using Arduino 1.8.3.

SvenDowideit, 2017-07-09:
That does a little. But the menus are still mostly unusable

cmaglie, 2017-07-14:
Does **GNOME** provides a setting for hi-dpi display? How is it set?

SvenDowideit, 2017-07-31:
@cmaglie not that we found - GNOME's claim to fame is to remove settings that users use - and even the secret settings tool didn't help.

JamaicanMoose, 2017-11-02:
[image]
Not sure if this is the right place to post this, but on Ubuntu 17.10 using wayland, manual High DPI scaling works for everything except for the menu items. This isn't an issue that I have with other applications.

stapelberg, 2019-09-07:
As a shot into the dark: check your `gtk-font-name` value. I had `gtk-font-name "Bitstream Vera Sans 8"` in my `~/.gtkrc-2.0`. Changing font size from 8 to 12 worked for me.

Gaara14, 2019-11-24:
In addition, it's possible to modify the menu font size:
At the beginning of ./arduino script, just add
`xrandr --dpi xx`
with xx the value needed. In my case, it's 122.
In the above script you can reset `xrandr` back to default DPI of 96 by adding the following line to the bottom:
`xrandr --dpi 96`

s-light, 2020-02-07:
as @stapelberg mentioned - his a workaround works ;-)  
i posted a detailed way in [#9170](https://github.com/arduino/Arduino/issues/9170#issuecomment-583397780)
basically create a custom `.gtkrc-2.0` file with font size you like and start arduino with this file:
`GTK2_RC_FILES=my_arduino_gtkrc-2.0.conf arduino`

cacois, 2020-07-20:
Still seeing this problem in PopOS 20.04 (ubuntu 20.04 base). Weird highlighting of text on mouseover, clicks can't register because everything is off in location.

Its basically unusable :(

dpellegr, 2020-09-04:
Do the following.

1. Find the Arduino launch script, it could simply be `/usr/bin/arduino` or some other file called therein, for instance Arch has it at this path `/usr/share/arduino/arduino`.
2. Find the line which actually launches java. For Arch this is the very last line in the file and it looks like this:
```"$JAVA" "${JAVA_OPTIONS[@]}" processing.app.Base "$@"```
3. Add the new line:
```JAVA_OPTIONS+=("-Dsun.java2d.uiScale=2")```
 right before the line at point 2.

Everything (text, menu, icons...) will be scaled by a factor 2. You can tune the factor to your screen/taste, however only integers seems to be accepted.

For more details: https://wiki.archlinux.org/index.php/HiDPI#AWT/Swing

cacois, 2020-09-04:
The solution for me was to uninstall the flatpak (How the GUI package installer wanted to install things), and install Arduino IDE from the tarball directly according to: [https://www.arduino.cc/en/guide/linux](https://www.arduino.cc/en/guide/linux). This installed fine, and works perfectly on high DPI. No idea what was wrong with the flatpak install.

phrxmd, 2020-12-16:
[image]

This is Arduino IDE 1.8.13 on OpenSUSE  TW, installed from the distribution package, on X11, on a Thinkpad X1 with resolution 2560x1440, KDE display scaling set to 200%, and an empty `~/.arduino15/preferences.txt` (Arduino default scaling preferences: scale 100% and autodetect enabled). Arduino IDE interface elements are unusably large, and its windows (such as the preferences window and library) are unusable.

If I disable autoscaling in the Arduino IDE and set scale to 50%, I get reasonably-sized text in the editor, but all UI elements are still enormous, their size doesn't seem to be influenced at all by anything I do in the Arduino preferences.

I wish in 2020 these things were somehow under control. I would switch to Wayland, but it doesn't support some other things I need.

techtasie, 2021-01-09:
If all the tips above still not working for you just try using the flatpack version instead of the apt version. [Flatpack](https://flathub.org/apps/details/cc.arduino.arduinoide)

KaiMartin, 2023-04-27:
Like others have reported, the preferences from within the GUI did not affect the menu for me (Debian bookworm/testing).  Settings in gtkrc-2.0 did not do anything.
Turns out, that recent versions of java switched to GTK3 for their GUI. What finally worked for me, was to set the shell variable GDK_SCALE:
    export GDK_SCALE=2 
(Took the hint from https://wiki.archlinux.org/title/HiDPI#GUI_toolkits )

This is still a dirty hack. Icons and the corresponding tool tips render a bit mushy. This is to be expected, as each of their pixels turns into a tiny 2x2 square. However, the fonts in the main UI and in the menu render fine and crisp. I get a decent UI on a 4k, 24" monitor.

[image]