arduino/Arduino #6474: builtin_tools_versions.txt file in the tool of a hardware package breaks compilations in other hardware packages

per1234, 2017-07-09:
Arduino IDE 1.8.3, Windows 7 64 bit

Steps to reproduce (only set up for Windows):

1. File > Preferences > Additional Boards Manager URLs > https://raw.githubusercontent.com/per1234/misc/issue_6473/package_issue_6473_index.json > OK
2. Tools > Board > Boards Manager
3. Wait for downloads to complete
4. Click on "Arduino SAMD Boards" and click "Install" (if you don't already have it installed)
5. Wait for installation to complete
6. Scroll down to "issue_6473", click on it.
7. Click "Install"
8. Wait for installation to complete
9. Click "Close"
10. Tools > Board > Arduino/Genuino Zero
11. Sketch > Compile/verify - Compilation fails with `fatal error: sam.h: No such file or directory`
12. Tools > Board > issue_6473
13. Sketch > Compile/verify - Compilation is successful
14. Delete Arduino15/packages/issue_6473/tools/avr-gcc-elektor/4.9.2-elektor/avr/builtin_tools_versions.txt
10. Tools > Board > Arduino/Genuino Zero
15. Sketch > Compile/verify - Compilation is successful

The same breakage occurs for Intel Galileo:
```
exec: "/pokysdk/usr/bin/i586-poky-linux-uclibc/i586-poky-linux-uclibc-g++": file does not exist
```
and Intel Edison:
```
exec: "/pokysdk/usr/bin/i586-poky-linux/i586-poky-linux-g++": file does not exist
```

Since the hardware package that causes the issue works fine and the problem manifests itself in a different hardware package than the actual source of the bug this will lead to support requests to the wrong place, most likely Arduino.

This bug was originally discovered in https://github.com/arduino/Arduino/issues/4593#issuecomment-187003469. I'm creating a new issue because that issue is not very concise due to the cause only being discovered after the issue was submitted. Also it is no longer possible to easily reproduce the issue following the information in that issue since the hardware package was fixed. Interestingly, the issue manifests itself in a different manner in Arduino IDE 1.6.7, the latest release at that time, breaking Arduino 101 (which is not affected in Arduino IDE 1.8.3) as well as some other boards.

facchinm, 2017-09-11:
Hi @per1234 ,
I believe this issue should be moved to https://github.com/arduino/arduino-builder since that file is only considered if contained in `bundledHardware` folder https://github.com/arduino/Arduino/blob/master/arduino-core/src/cc/arduino/contributions/packages/ContributionsIndexer.java#L216 .

Instead, in the builder, we read it for ANY hardware folder (https://github.com/arduino/arduino-builder/blob/a07ea16effee9c7402f877c063ac1ac4702b856d/src/arduino.cc/builder/tools_loader.go#L149).

Theoretically, everything should work just fine if we remove the builtin-stuff from the builder, since we pass the runtime.* variables (https://github.com/arduino/Arduino/commit/6f24fa6cecd9d0e3a650ff8ed109cbc121c1ceab), but this would harm its "drop-in" compatibility.

I'd rather prefer to emit a warning if multiple `builtin_tools_versions.txt` are found; what do you think about it?

per1234, 2017-09-11:
Sorry for posting to the wrong repository. I have moved it to https://github.com/arduino/arduino-builder/issues/246.

closed 2017-09-11 by per1234