arduino/Arduino #6475: Add support to --version and --help from the command line

simonepri, 2017-07-10:
The command line support was add from the version 1.5.2 of the IDE but some basics commands are missing:
https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc

A great start could be to add those flags:

Flag | Description
-----|------------
--version | Shows the version of the binary
--help     | Shows the list of commands available

Currently I'm trying to develop cli wrappers for the Arduino binary in different languages (Currently only in node, but not published yet) in order to make the Arduino cross-platform compilation simpler and less stressful.
https://github.com/arduino-cli

per1234, 2017-07-10:
Closing as duplicate of https://github.com/arduino/Arduino/issues/5098

We welcome bug reports and feature requests but in the future please take the time to search for existing open and closed issues and pull requests before submitting an issue:
https://github.com/arduino/Arduino/issues?&q=

closed 2017-07-10 by per1234

simonepri, 2017-07-10:
Sorry about that, I didn't see it.