arduino/Arduino #6483: Mislabeled Perfboard Power Rails

grayl, 2017-07-13:
In the perfboard examples in https://www.arduino.cc/en/Tutorial/ArduinoISP, the power rails are not correctly labeled. They should be paired top/top, bottom/bottom, not top/bottom, bottom/top

per1234, 2017-07-13:
Just to make the issue more clear, even though the polarity of the power rails on the tutorial page are not marked, in the real world every breadboard of this type I've seen had the polarity labeled in a way that contradicts the diagram in the tutorial. For example, https://www.adafruit.com/product/64:
[image]

SimonePDA, 2018-03-06:
THe Arduino as ISP has been completely revamped and the issue is now obsolete.

closed 2018-03-06 by SimonePDA