arduino/Arduino #6490: per project settings

jonathan-annett, 2017-07-14:
Whilst I appreciate that the current system is fine for beginners who work on one project at a time, you really need to address the cluster-f#&k centered around configuration. if one wants to work on code that (for example) establishes communication between 2 boards, it rapidly becomes an issue. even if they are identical boards running different code with identical build settings but are connected on a different usb port, it's very easy to send the wrong code to the wrong port.

so here's what i suggest 

1) add a checkbox in the configuration that enables "per project settings". leaving this unchecked (the default) would not change anything about how configuration is handled, meaning it's backwardly compatiable for users who want to keep the status quo, or learners who find the idea of per project settings daunting.

2) with the checkbox checked, all settings that affect how a project is built or uploaded are saved along with the project. this includes the port being used, the board type, library setttings, weather you have the terminal window open, etc etc.

3) it's fair enough that you only want one compile session happening at a time. but for goodness sake, if you start compiling a project then switch focus in the ide to another project, don't just start logging build messages to the new project window. this is stupid. keep the messages in the project window they belong to, and just disable the build/upload button for any other project. this means you can go and edit some other project while one  is being built, and return to the presumably built project later to see how it's is getting on.

4) terminal windows. there is no sane reason you can't have multiple terminal windows open at once, and i'd suggest you allow them to be docked to the project window they are associated with. this means if you are debugging code that deals with 2 boards at once, you don't need to resort to third party terminal windows to see the serial output of both boards at once. 
5) a nice to have power user feature would be to allow the terminal window output to be merged into one window with different colours per serial port they are connected to. obviously any input needs to be disabled, or clearly indicated which serial port you are sending to if you happen to start typing, but let's face it, most users use the terminal to display logging info, and ignore the fact they can type stuff into the terminal. so why not just have a keyboard mute button, which can be optionally assigned to send to a specific board if that's what they want.

ok implementing all of these changes will require a bit of work, but the first few (and fixing the compiler focus issue) should fairly trivial, and will hopefully stem the tide of users abandoning your ide in favour of other more sanely managed solutions.

per1234, 2017-07-14:
> 1. add a checkbox in the configuration that enables "per project settings". leaving this unchecked (the default) would not change anything about how configuration is handled, meaning it's backwardly [...]

Duplicate of https://github.com/arduino/Arduino/issues/1481 (and note that some of this functionality is available when using the [Arduino Web Editor](https://create.arduino.cc/editor)).

> 3. it's fair enough that you only want one compile session happening at a time. but for goodness sake, if you start compiling a project then switch focus in the ide to another project, don't just [...]

Duplicate of https://github.com/arduino/Arduino/issues/2680

AFAIK 4 and 5 are original feature requests. In the Arduino world what you refer to as a "terminal window" is called the Serial Monitor.

We welcome bug reports and feature requests but in the future please take the time to search for existing open and closed issues and pull requests before submitting an issue:
https://github.com/arduino/Arduino/issues?&q=

Please keep each issue to a single topic. Otherwise it is more difficult to label, assign, find in search results, and track their progress and the discussion is likely to get fragmented.

>fixing the compiler focus issue) should fairly trivial

See the comments in https://github.com/arduino/Arduino/issues/2680. It sounds like a pull request for this fix would be welcome.

okpapt, 2017-10-12:
You can open several instance of IDE Arduino by double click on arduino.exe, each have a separated serail port and monitor.  After open file.ino in each instance of IDE Arduino.

per1234, 2022-12-01:
>terminal windows. there is no sane reason you can't have multiple terminal windows open at once

Full support (without the multiple instances hack required by Arduino IDE 1.x) for selecting a different board+port in each Arduino IDE window and having a Serial Monitor open on each has been added in Arduino IDE 2.x.

> i'd suggest you allow them to be docked to the project window

Serial Monitor is currently part of the Arduino IDE 2.x window. The ability to make this optional is tracked at https://github.com/arduino/arduino-ide/issues/289

>allow the terminal window output to be merged into one window

We are declining to add this feature. It will be best to simply use separate Serial Monitors for each port.

closed 2022-12-01 by per1234