arduino/Arduino #6491: IF content is done in half. It should not at all.

lunter2, 2017-07-15:
void setup() {
  byte pin = 7;

  pinMode(pin, INPUT_PULLUP);

  byte a;
  byte b = 0;
  
  if (digitalRead(pin) == LOW) { // condition always false (no GND on pin)
   a = 123;
   b = 234;
  }

  Serial.begin(9600);
  Serial.println(a); // 123 (should be 0)
  Serial.println(b); // 0
}

void loop() {

}

per1234, 2017-07-15:
Uninitialized local variables may contain any value. Thus you have no assurance that `a` will be 0 in your code if `pin` is `HIGH`. Try this:
```arduino
void setup() {
  byte pin = 7;
  pinMode(pin, INPUT_PULLUP);
  byte a = 0;
  byte b = 0;
  if (digitalRead(pin) == LOW) { // condition always false (no GND on pin)
    a = 123;
    b = 234;
  }
  Serial.begin(9600);
  Serial.println(a); // 123 (should be 0)
  Serial.println(b); // 0
}

void loop() {}
```

In the future please use code fencing when you post code to the issue tracker, as specified in the [Contributing Guide](https://github.com/arduino/Arduino/blob/master/CONTRIBUTING.md).

In the future please add a descriptive title to your issue reports.

closed 2017-07-15 by per1234

lunter2, 2017-07-15:
>> Uninitialized local variables may contain any value.

But why any is always 123 ?

lunter2, 2017-07-15:
```
void setup() {
  Serial.begin(9600);

  byte pin = 7;

  pinMode(pin, INPUT_PULLUP);

  byte a;
  byte b = 0;
  Serial.println(a); // 0
  Serial.println(b); // 0

  a++;
  b++;
  Serial.println(a); // 1
  Serial.println(b); // 1

  if (digitalRead(pin) == LOW) { // condition always false (no GND on pin)
   a = 123;
   b = 234;
  }

  Serial.println(a); // 123 (should be 1)
  Serial.println(b); // 1
}

void loop() {

}
```

Try this.
a is not uninitialized local variable because first Serial.println(a) prints 0 and second 1.

cmaglie, 2017-07-15:
Accessing an uninitialized variable is **undefined** behaviour in C++. "Undefined" means that literally everything can happen: the program may crash or show weird outputs like two consecutive `println(a)` that prints different values.

A thing that may cause confusion is that *some kind of* variables may be automatic initialized to `0` (for example global variables) in that case the explicit initialization is not needed. BTW the rules to determine which variable is automatically initialized in C++ are really complex, so it's a good rule of thumb is to always initialize variables explicitly.

PaulStoffregen, 2017-07-15:
> But why any is always 123 ?

Because your assumption that the pin is never low is **wrong**.  The pin actually is low.

The input pullup resistors are very weak.  The pins have a small capacitance.  The pin does not actually become high until a short time after you use pinMode, because the weak pullups simply take time to charge the capacitance of the pin (plus any wire or other stuff connected).

If you add a delay before the digitalRead, *and* if you run this code after the board has been powered down for some time (so any previous write of 123 to that memory location is lost), you should have a 1-in-256 chance of seeing 123.  But it really depends on whatever random data is in the memory.  The memory can sometimes retain data (but not very reliably) for many seconds after powering down, so if you test this, keep the power off for quite some time to "lose" the prior 123 in memory.