arduino/Arduino #6503: Example code on reference pages do not follow the Arduino code format standard

per1234, 2017-07-16:
**Tools > Auto Format** in the Arduino IDE has established a standard of formatting for all Arduino sketches. It's important to teach beginners to be consistent in whatever code formatting style they choose. Unfortunately Arduino's example code doesn't always follow their own standard. I recommend doing a **Tools > Auto Format** style formatting on all example code in the Arduino reference pages. If it's more convenient, this can be done directly using the [AStyle tool](http://astyle.sourceforge.net/), using the settings in Arduino's [formatter.conf](https://github.com/arduino/Arduino/blob/1.8.3/build/shared/lib/formatter.conf).

kengdahl, 2018-03-22:
Hey. Could you give me an example of this from the examples?

per1234, 2018-03-22:
Since it's now possible for anyone to improve the Language Reference pages I did a quick browse of some of the reference pages for the more popular Arduino libraries and found these (not a comprehensive list):

- https://www.arduino.cc/en/Reference/EthernetIPAddress?action=sourceblock&num=1
- https://www.arduino.cc/en/Reference/EthernetServer?action=sourceblock&num=1
- https://www.arduino.cc/en/Reference/EthernetUDPBegin?action=sourceblock&num=1
- https://www.arduino.cc/en/Reference/EthernetUDPRead?action=sourceblock&num=1
- https://www.arduino.cc/en/Reference/EthernetUDPBeginPacket?action=sourceblock&num=1
- https://www.arduino.cc/en/Reference/EthernetUDPEndPacket?action=sourceblock&num=1
- https://www.arduino.cc/en/Reference/EthernetUDPParsePacket?action=sourceblock&num=1
- https://www.arduino.cc/en/Reference/EthernetUDPAvailable?action=sourceblock&num=1
- https://www.arduino.cc/en/Reference/EthernetUDPRemoteIP?action=sourceblock&num=1
- https://www.arduino.cc/en/Reference/EthernetUDPRemotePort?action=sourceblock&num=1
- https://www.arduino.cc/en/Reference/FileIsDirectory?action=sourceblock&num=1
- https://www.arduino.cc/en/Reference/DueExtendedSPI?action=sourceblock&num=4
- https://www.arduino.cc/en/Reference/DueExtendedSPI?action=sourceblock&num=3
- https://www.arduino.cc/en/Reference/SoftwareSerialAvailable?action=sourceblock&num=1
- https://www.arduino.cc/en/Reference/SoftwareSerialIsListening?action=sourceblock&num=1
- https://www.arduino.cc/en/Reference/SoftwareSerialOverflow?action=sourceblock&num=1
- https://www.arduino.cc/en/Reference/SoftwareSerialPeek?action=sourceblock&num=1
- https://www.arduino.cc/en/Reference/SoftwareSerialRead?action=sourceblock&num=1
- https://www.arduino.cc/en/Reference/SoftwareSerialPrint?action=sourceblock&num=1
- https://www.arduino.cc/en/Reference/SoftwareSerialPrintln?action=sourceblock&num=1
- https://www.arduino.cc/en/Reference/SoftwareSerialListen?action=sourceblock&num=1
- https://www.arduino.cc/en/Reference/SoftwareSerialWrite?action=sourceblock&num=1

Although not enforced by Auto Format, I also find it quite unfortunate that there is no consistent brace style (broken/attached) anywhere in Arduino code. If a specific style convention was established then it would be easy to conform to. As things are now, I've wasted time trying to figure out which style is used in the majority of official Arduino code.