arduino/Arduino #6505: Incorrect parameter type in digitalRead() reference page

per1234, 2017-07-16:
https://www.arduino.cc/en/Reference/DigitalRead
>Parameters
pin: the number of the digital pin you want to read (int) 

Actually the `pin` parameter is uint8_t (from https://github.com/arduino/Arduino/blob/1.8.3/hardware/arduino/avr/cores/arduino/Arduino.h#L135):
```
int digitalRead(uint8_t);
```
but maybe it's best to call it "byte" in the reference page.

oqibidipo, 2017-07-16:
> Actually the pin parameter is uint8_t 

... except in SAM and SAMD cores it is uint32_t.

per1234, 2017-11-14:
The arduino/reference-en version of the reference page no longer documents the parameter type (which I must admit was [my doing](https://github.com/arduino/reference-en/commit/d21b22c1a144c467cd3308bd47f148b2c1626b0b)). That was not at all a good solution to this issue but it does mean this can now be closed in favor of the more general https://github.com/arduino/reference-en/issues/15.

closed 2017-11-14 by per1234