arduino/Arduino #6510: Issues with Arduino USB Host Shield product page

per1234, 2017-07-16:
https://www.arduino.cc/en/Main/ArduinoUSBHostShield

Change:
>The Arduino USB Host Shield allows you to connect an USB device to your Arduino.

to:
>The Arduino USB Host Shield allows you to connect **a** USB device to your Arduino.

I suppose you could make an argument for either but there are two instances of "a USB" in the rest of the page and no other instances of "an USB" and I prefer "a USB".

----
## OVERVIEW tab
Change:
>The shield isTinkerKit compatible

to:
>The shield **is TinkerKit** compatible

----
>- [Arduino's ADK documentation.](http://arduino.cc/en/Reference/AndroidAccessory)

This was reported years ago in https://github.com/arduino/Arduino/issues/2410 and closed "since the ADK hardware is marked as retired and the documentation will not be online any more". At the very least the link should be removed from the page.

Also, what about this: https://www.arduino.cc/en/Guide/ArduinoADK?

----
Change:
>You can find here your board warranty informations.

to:
>You can find here your board warranty **information**.

I suspect every product page has this typo

----
Change:
>You can find in the Getting Started section all the information you need to configure your board, use the Arduino So ftware (IDE), and start tinker with coding and electronics.

to:
>In the Getting Started section you can find all the information you need to configure your board

----
Change:
>use the Arduino So ftware (IDE), and start tinker with coding and electronics.

to:
>use the Arduino **Software** (IDE), and start tinker with coding and electronics.

----
>On the Software [on the Arduino Forum](https://forum.arduino.cc/index.php?board=63.0)

This is a link to the old, read-only forum. I don't know where this link should go but that's definitely wrong.

----
Change:
>On the Product itself through [our Customer Support](https://store.arduino.cc/index.php?main_page=contact_us&language=en)

to:
>On the Product itself through [our Customer Support](https://store.arduino.cc/store-support)

The previous link just dumps me in the Arduino Store home page. I suspect all product pages have this bug.

## TECH SPECS tab
As [specified by SI](http://www.bipm.org/en/publications/si-brochure/section5-3.html#section3):
>a space is always used to separate the unit from the number

This is done inconsistently in the page. Thus "5V" should be written as "5 V", "500mA" as "500 mA", "400mA" as "400 mA".

## DOCUMENTATION tab
Change:
>Kristian Lauszus and Andrew Kroll onGitHub (download).

to:
>Kristian Lauszus and Andrew Kroll **on GitHub** (download).

----
Change:
>With an Arduino HSB Host Shield you can even [control an RC car using an XBOX wireless game controller](http://scuola.arduino.cc/courses/lessons/cover/wXd2go1).

to:
>With an Arduino HSB Host Shield you can even [control an RC car using an XBOX wireless game controller](https://create.arduino.cc/projecthub/Arduino_Scuola/joystick-controlled-rc-car-076b7f)

(fix the dead link)

----
Change:
>here you can find some tutorial

to:
>here you can find some **tutorials**

----
Change:
>developed bycircuits@home.

to:
>developed **by circuits@home.**

----
Change:
>- Bluetooth HID devices;
>- PS4 controller;
>-  Adding a display to a digital scale using Arduino and USB Host shield;
>- Mass Storage;

to:
>- Bluetooth HID devices
>- PS4 controller
>-  Adding a display to a digital scale using Arduino and USB Host shield
>- Mass Storage
 
----
Change:
>This shield has a MALE USB port

to:
>This shield has a Type-A USB receptacle

kengdahl, 2018-03-21:
I changed the Arduino's ADK documentation link to point to this:
https://www.arduino.cc/en/Guide/ArduinoADK

kengdahl, 2018-03-21:
I think I've fixed everything now.

closed 2018-03-21 by kengdahl