arduino/Arduino #6514: Typos in "Shipping policy" page

per1234, 2017-07-16:
https://store.arduino.cc/usa/shipping-policy

Change:
>Carrier delivery time does not include order processing period as mention in next section

to:
>Carrier delivery time does not include order processing period as **mentioned** in **the** next section

----
Change:
>BCMI US LLC cannot be hold responsible for the Carriers late deliveries.

to:
>BCMI US LLC cannot be **held** responsible for the **Carrier's** late deliveries.

----
Change:
>can be subject to custom controls for duties and vat for import. 

to:
>can be subject to custom controls for duties and VAT for import.

----
Change:
>BCMI US LLC cannot be hold responsible for the late deliveries due to custom control.

to:
>BCMI US LLC cannot be **held** responsible for late deliveries due to customs control.

----
Change:
>Time requirements are meant working days only

to:
>Time requirements refer to business days only

----
Change:
>monday to friday

to:
>Monday to Friday

----
Change:
>national holiday excluded

to:
>national **holidays** excluded

----
Change:
>Orders entered before 1:30 pm EDT (GMT-4) Mon-Fri Best Effort ship same day while are Guaranteed Next Business Day

to:
>Orders entered before 1:30 pm EDT (GMT-4) Mon-Fri **we make our best effort to ship same day and  guarantee next business day shipping**

----
Change:
>Orders entered before 4:30 pm EDT (GMT-4) Mon-Sun ship Next Business Day

to:
>Orders entered before 4:30 pm EDT (GMT-4) Mon-Sun ship **next business day**

kengdahl, 2017-09-27:
These issues are fixed now. Thank you @per1234 , for the headsup

per1234, 2017-09-27:
@kengdahl the page remains unchanged. Did your fixes maybe not get pushed to the public server?

kengdahl, 2017-12-21:
Ok. the US page is also updated now.

per1234, 2018-01-15:
@kengdahl there was one item missed:
Change:
>Carrier delivery time does not include order processing period as mention in next section

to:
>Carrier delivery time does not include order processing period as **mentioned** in **the** next section

kengdahl, 2018-01-31:
**Fixed**

closed 2018-01-31 by kengdahl