arduino/Arduino #6515: Typos in

per1234, 2017-07-16:


kengdahl, 2017-09-26:
Thank you @per1234 ! It is now fixed with your suggested fixes.

per1234, 2017-09-27:
@kengdahl, the page is unchanged. Maybe your fixes weren't pushed to the public server?

kengdahl, 2017-09-28:
Is it still iunchanged? @per1234

per1234, 2017-09-28:
When I load https://store.arduino.cc/usa/payment-methods it's still unchanged. I'm using ctrl+F5 to do a "Reload (override cache)".

When I change my location to Europe/Africa and load https://store.arduino.cc/payment-methods I do see the changes. On that page I notice an issue:

>Arduino Store is accepts payments via Credit / Debit Cards and PayPal

should be:
>Arduino Store accepts payments via Credit / Debit Cards and PayPal

kengdahl, 2017-12-21:
EU and US page are now fixed.

per1234, 2018-01-15:
@kengdahl I accidentally introduced a typo with my suggestion:
Change:
>In case of potential fraud or misuse, we may chose to not ship an order, or request additional verification information.

to:
>In case of potential fraud or misuse, we may **choose** to not ship an order, or request additional verification information.

Sorry about that!

kengdahl, 2018-01-31:
**Fixed**

closed 2018-01-31 by kengdahl