arduino/Arduino #6516: Bad URL in "SELLER TERMS" page

per1234, 2017-07-16:
https://store.arduino.cc/usa/seller-terms

Under the "12. SUPPORT." section:
>Customer shall contact the Arduino Support team via the web form located at http://store-usa.arduino.cc/pages/contact-us-store-support.

That URL just dumps me to the Arduino Store home page. The correct link is:
https://store.arduino.cc/usa/store-support

The same problem occurs on the product pages so I don't know whether it's better to change the link or to make it so that http://store-usa.arduino.cc/pages/contact-us-store-support will actually work.

Also, it seems like it would be a good idea to make that URL clickable to make it as easy as possible for customers who are having a "Problem".

Note that the "[please fill this form](https://store.arduino.cc/usa/store-support)" link in the "25. QUESTIONS" at the bottom of the page does work correctly.

mastrolinux, 2017-09-28:
Hi,

I do not understand, all links work here, can you please make a video or a couple of images?

per1234, 2017-09-28:
The redirect from http://store-usa.arduino.cc/pages/contact-us-store-support to https://store.arduino.cc/usa/store-support is working correctly for me now.

The issue of the link not being clickable remains.

kengdahl, 2017-12-21:
The link is now clickable.

closed 2017-12-21 by kengdahl