arduino/Arduino #6518: Typos on "Transparency

per1234, 2017-07-16:
>you can check every details

to:
>you can check every **detail**

kengdahl, 2017-09-28:
**Only these two left: **
----
The "PAYMENTS" section seems to result in duplicate content with https://store.arduino.cc/usa/payment-methods. You might consider migrating unique content to that page and then just having a link to it from the Transparency page.

----
The "SHIPMENTS" section seems to result in duplicate content with https://store.arduino.cc/usa/shipping-policy. You might consider migrating unique content to that page and then just having a link to it from the Transparency  page.

----

per1234, 2018-01-15:
@kengdahl I spotted a newly added typo.
Change:
>Please contact us thruu our Store support form.

to:
>Please contact us **through** our Store support form.

kengdahl, 2018-01-31:
**Fixed**

Was only left on the US page. Must have forgot to fix it.

closed 2018-01-31 by kengdahl