arduino/Arduino #6519: Typos on "Educators" page

per1234, 2017-07-16:
https://store.arduino.cc/usa/educators

----
Change:
>How do I change items' quantities in my cart?

to:
>How do I change **item** quantities in my cart?

(two instances)

----
Change:
>The Checkout is a 3 steps process

to:
>The Checkout is a 3 **step** process

----
Change:
>We no longer not accept email or fax copies of the PO

to:
>We no longer accept email or fax copies of the PO

----
Change:
>Our pricing and availability information is online on store.areduino.cc

to:
>Our pricing and availability information is online on **store.arduino.cc**

Also: Might be a good idea to make that URL clickable

----
Change:
>orders will have to be approved manually from one of our representatives

to:
>orders will have to be approved manually **by** one of our representatives

----
Change:
>Please contact us as soon as possible in case you wish your order to be expedited

to:
>Please contact us as soon as possible **if** you wish your order to be expedited

----
>Get in touch with our Support team support@arduino.cc in order to determine the ongoing issue.

The other email links are clickable but this one is not.

kengdahl, 2017-09-26:
Thanks for the headsup! These are fixed now

per1234, 2017-09-27:
@kengdahl, the page is unchanged. Maybe your fixes weren't pushed to the public server?

kengdahl, 2017-09-28:
Actually it's the US version of the store that is separate from the rest of the world. I am currently working on fixing that one too.

kengdahl, 2017-12-21:
This should be fixed now. Please have a look.

closed 2017-12-21 by kengdahl