arduino/Arduino #6520: Typos on AtHeart page

per1234, 2017-07-16:
https://www.arduino.cc/en/ArduinoAtHeart/Products

----
The "Shop AtHeart products on our Store." link points to:
https://store.arduino.cc/category/84
but that just dumps me to the Arduino Store homepage.

The correct link (for USA region) is:
https://store.arduino.cc/usa/atheart

----
Change:
>In the heart of MicroView there is an ATMEL’s ATmega328P

to:
>In the heart of MicroView there is an Atmel ATmega328P

----
www.microview.io gives a "Server not found" error

----
Change:
>- ATMega328 clocked at 8 or 16 MHz
>- ATMega1280 clocked at 16 MHz
>- ATMega2560 clocked at 16 MHz
>- ATMega32U4 at 8 or 16 MHz	
>- ATMega64M1 at 16 Mhz 

to:
>- ATmega328P clocked at 8 or 16 MHz
>- ATmega1280 clocked at 16 MHz
>- ATmega2560 clocked at 16 MHz
>- ATmega32U4 at 8 or 16 MHz	
>- ATmega64M1 at 16 Mhz 

Also: what about ATmega168 and ATtiny85 (or ATmega8 if you want to get really complete)?
----
For more information about the AtHeart program please contact at_heart [at] arduino.cc

The non-obfuscated email address is shown at the top of the page so I don't think this has any value for spam prevention.

kengdahl, 2017-09-13:
Thank you @per1234 

Fixed.

per1234, 2017-09-13:
@kengdahl, the link [Shop AtHeart products on our Store.](https://store.arduino.cc/atheart), although updated, still dumps me in the store home page. https://store.arduino.cc/usa/atheart does take me to the right store section. I would expect the store to be able to take the non-regional URL and convert it to the correct regional URL, as it does for product pages (e.g. https://store.arduino.cc/arduino-uno-rev3 redirects to https://store.arduino.cc/usa/arduino-uno-rev3) but apparently this doesn't work for store sections. Maybe that's a separate issue that should be reported and the link on the AtHeart page can be left as-is in the hopes that it will eventually work?

kengdahl, 2017-09-13:
I'm reopening the issue. I'll look further into this with our main admin.

kengdahl, 2017-12-21:
The link to the atheart products should work now. Better late than never, right :)

closed 2017-12-21 by kengdahl