arduino/Arduino #6523: Suggested improvements for "Contact Us" page

per1234, 2017-07-16:
https://www.arduino.cc/en/Main/ContactUs

----
Change:
>If you're not sure which is the appropriate board, use the “project guidance” section:

to:
>If you're not sure which is the appropriate board, use the “**Project Guidance**” section:

----
Change:
>check if it's already known by looking at http://github.com/arduino/Arduino/issues

to:
>check if it's already known by looking at **https://github.com/arduino/Arduino/issues?q=**

This change is necessary because people should search through all open and closed issues and pull requests. The previous link will only display open issues.

----
Change:
>and if it's not known, please [add it to the list](https://github.com/arduino/Arduino/issues?q=is%3Aopen+sort%3Acreated-desc+label%3A%22Component%3A+Documentation%22). 

to:
>and if it's not known, please [**add it to the list**](https://github.com/arduino/Arduino/issues?label%3A%22Component%3A+Documentation%22)

(link change) This change is necessary because people should search through all open and closed issues. The previous link will only display open issues.

----
Change:
>If you are looking to help with translation of the Arduino IDE, please use Transifex :

to:
>If you are looking to help with translation of the Arduino IDE, please use **Transifex:**

kengdahl, 2017-09-12:
Fixed.

closed 2017-09-12 by kengdahl

per1234, 2017-09-12:
Thanks @kengdahl. The last two suggestions were not implemented. Of course this is completely up to you but I just wanted to make sure that was intentional.