arduino/Arduino #6526: Incorrect link for corrections/suggestions on all documentation pages

per1234, 2017-07-17:
At the bottom of all documentation pages is this line:
>Corrections, suggestions, and new documentation should be posted to the [Forum](http://arduino.cc/forum/index.php/board,23.0.html).

The link points to the "Other Software Development" section of the forum. The "[Suggestions for the Arduino Project](http://forum.arduino.cc/index.php?board=21.0)" section seems much more appropriate.

kengdahl, 2018-03-27:
Are you refering to the documentation tabs in the store?

per1234, 2021-04-19:
... and now after the forum update that link just takes you to an "Oops! That page doesn’t exist or is private." page.

per1234, 2022-07-17:
It was fixed at some point but nobody bothered to close this.

closed 2022-07-17 by per1234