arduino/Arduino #6530: z Finnish translations in the IDE.

Katajisto, 2017-07-18:


closed 2017-07-18 by Katajisto

per1234, 2017-07-18:
@Katajisto I see you already closed this but if anyone is interested in helping with translations of the Arduino IDE it would be much appreciated. You can get involved here:
https://www.transifex.com/mbanzi/arduino-ide-15/
It says the Finnish translation is currently 61% complete with 226 strings left to translate.

Katajisto, 2017-07-18:
Hey,
I started typing the issue, and accidentally pressed enter before finishing the post,
so I tought I'd just close it and open a new issue tomorrow.

Thanks for the link. As a freelance translator some of the absolutely horrible translations really bothered me, for example the finnish word "sketsi" that is used for "sketch" actually means "sketch comedy". I am going to visit that site and make it go to 100% and also fix the existing ones.