arduino/Arduino #6536: Serial Monitor ASCII Tab Displays Inconsistently

SolidHal, 2017-07-20:
When printing the tab character from my teensy 2.0 to the serial monitor, it outputs multiple different sizes of tab. It seems to be related to the number of characters before the tab. 
Here is the test code:

`
boolean NotRun = true;
void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  Serial.println("Beginning");
}

void loop() {
  // put your main code here, to run repeatedly:
  if(NotRun){
    Serial.print('g');
    Serial.print('\t');
    Serial.print('g');
    Serial.print('g');
    Serial.print('g');
    Serial.print('g');
    Serial.print('\t');
    Serial.print('g');
    Serial.print('\t');
    Serial.print('\t');
    Serial.print('\t');
    Serial.print('g');
    NotRun = false;
    Serial.println();
  }
}`

I've attached a screenshot of the serial monitor output. As you can see, the first tab printed is much longer than the second tab printed. The three tabs printed back to back each seem to be the same size as the first tab printed, as you can see from the highlighted tab in the second screenshot. This occurs with both the '/t' character and if the ASCII tab value 9 is converted to a character then printed. 

I am running Arduino 1.86 on a Mac, with teensyduino 1.36. 
I believe this is an issue with the way the serial monitor displays the tab character under certain conditions as when I copy both the long and the short tab character from the monitor to a test editor that doesn't convert tabs to spaces, like textedit, they both appear as the same length.

[image]
[image]

per1234, 2017-07-20:
Maybe this will make it clear the reason for the "inconsistency":
```arduino
void setup() {
  Serial.begin(9600);
  Serial.println("g\tgg\tggg\tgggg");
  Serial.println("gg\tggg\tgggg\tggggg");
  Serial.println("ggg\tgggg\tggggg\tgggggg");
}
void loop() {}
```
```
g	gg	ggg	gggg
gg	ggg	gggg	ggggg
ggg	gggg	ggggg	gggggg
```
You can see that if all the tabs were a fixed number of spaces the columns wouldn't line up nicely.

matthijskooijman, 2017-07-20:
Yup, so this is by design. It is also how many text editors and terminals work. I'm closing this issue.

closed 2017-07-20 by matthijskooijman