arduino/Arduino #6537: I demand linumber! :)

RichardNg, 2017-07-20:
On my Mac Arduino IDE (mine is 1.6.10) I can simply go to the Arduino->Preferences check the display line number box.

But the linux version, not so. I am now trying to figure out how to set it.

A lot of the IDE, xCode included, don't have the line number set by default, some even omitted it from this option. I wonder why?  It seems to me it is shch as helpful feature in debugging code.

cmaglie, 2017-07-20:
The IDE is the **same** for MacOSX and Linux, maybe you just need to upgrade to the latest version...

(I'll tell you just in case: Ubuntu/debian have an ancient version of the IDE distributed from their package manager, so you need to uninstall that one and download from arduino.cc the latest)

RichardNg, 2017-07-20:
Oh I see .Thanks for the info. I happen to use Pi B install Raspbian
Jessie. May be that's why.

PaulStoffregen, 2017-07-20:
Almost the same.  I believe on Mac it's in the "Arduino" menu and on Linux & Windows you find Preferences under the "File" menu.

facchinm, 2017-07-25:
@RichardNg you can download the latest version of the IDE running on Raspberry from https://www.arduino.cc/download_handler.php?f=/arduino-1.8.3-linuxarm.tar.xz . As @cmaglie pointed out, most distributions ship an ancient version of the IDE and Raspbian is no exception :wink:

closed 2017-07-25 by facchinm

RichardNg, 2017-07-25:
Thank you so much. I installed the latest Linux version on my Pi as you
suggested, now I have the lovely line number just like the one I have on my
Mac. :)