arduino/Arduino #6539: [LIBRARY MANAGER]: Please add OpenBCI Libraries

aj-ptw, 2017-07-21:
Please add the following libraries

aj-ptw, 2017-07-21:
@per1234 could you also add

https://github.com/OpenBCI/OpenBCI_Radios

cmaglie, 2017-07-24:
There are a bunch of problems in your `library.properties`:

on https://github.com/OpenBCI/OpenBCI_Wifi_Master_Library and https://github.com/OpenBCI/OpenBCI_WIFI, you must define `author` and `maintainer` fields:

```
JOB 1041 - 2017/07/24 11:03:42 Checking out tag: v0.2.1
JOB 1041 - 2017/07/24 11:03:42 Error syncing library: Error generating library from repo: 'author' and 'maintainer' fields must be defined
```

on OpenBCI_Radios the name is set to `name=OpenBCI_` in library.properties:
https://github.com/OpenBCI/OpenBCI_Radios/blob/master/library.properties#L1
are you sure about that?

finally `architecutres="*"` (with quotes) will not work, you should change it to `architectures=*`, please change this on all your libraries.

aj-ptw, 2017-07-24:
@cmaglie could you please confirm the following libraries are all conforming?

https://github.com/OpenBCI/OpenBCI_Wifi_Master_Library/releases/tag/v0.0.5
https://github.com/OpenBCI/OpenBCI_WIFI/releases/tag/v0.2.2
https://github.com/OpenBCI/OpenBCI_Radios/releases/tag/v2.0.2

If there is more than one author or maintainer, then should the property field be plural?

Thanks!

AJ

cmaglie, 2017-07-25:
> @cmaglie could you please confirm the following libraries are all conforming?

> If there is more than one author or maintainer, then should the property field be plural?

looks ok now, except the `maintainers` field that must be singular even if there is more than one maintainer.

I've added all the libraries to the indexer:
- `OpenBCI_Wifi_Master` and `OpenBCI_Wifi` are ok they are already published.
- `OpenBCI_Radios` must be changed the field `maintainers` to `maintainer`. Once done the library will be automatically published within an hour, no need to do further actions here.

closed 2017-07-25 by cmaglie