arduino/Arduino #6540: Typos on Zero product page

per1234, 2017-07-21:
https://www.arduino.cc/en/Main/ArduinoBoardZero

----
Change:
>You can find your board warranty informations

to:
>You can find your board warranty **information**

----
>Arduino and Genuino boards based on AVR microcontrollers get a reset and restart sketch execution each time the Serial Monitor of the Arduino Software (IDE) is opened.

This is not true for the ATmega32U4 based Arduino boards.

----
Change:
>when the Zero is connected through the ProgrammingPort

to:
>when the Zero is connected through the **Programming Port**

----
Change:
>and you open the serial monitor

to:
>and you open the **Serial Monitor**

(consistent with the capitalization earlier in the paragraph)

----
Change:
>the Update Procedure explains what you should do each time there is a new Core release.

to:
>the Update Procedure explains what you should do each time there is a new **Arduino SAMD Boards** release.

----
Change:
>On the Product itself through [our Customer Support](https://store.arduino.cc/index.php?main_page=contact_us&language=en)

to:
>On the Product itself through [**our Customer Support**](https://store.arduino.cc/store-support)

(fix link)

----
As [specified by SI](http://www.bipm.org/en/publications/si-brochure/section5-3.html#section3):
>a space is always used to separate the unit from the number

This is done inconsistently in the page. Thus "5V" should be written as "5 V", "3.3V" as "3.3 V", "2.1mm" as "2.1 mm". The exception is when referring to pin names.

----
In "TECH SPECS" tab it says:
> PWM Pins   All but pins 2 and 7

but then in the "DOCUMENTATION" tab it says (twice):
>Pins that can be used for PWM output are: 3, 4, 5, 6, 8, 9, 10, 11, 12, 13 using analogWrite() function.

When I read "All" I expect that to mean all the pins, including 0, 1, and the analog pins.

----
Eagle file and schematics links are for MKRZero. I believe the correct links are:
- https://www.arduino.cc/en/uploads/Main/ArduinoZero-schematic.pdf
- https://www.arduino.cc/en/uploads/Main/ArduinoZero-reference.zip

(Originally reported at http://forum.arduino.cc/index.php?topic=490578)

----
Change:
>The Zero has been designed in collaboration with ATMEL

to:
>The Zero has been designed in collaboration with **Atmel**

----
Change:
>the on-board EDBG can be used through ATMEL Studio

to:
>the on-board EDBG can be used through **Atmel** Studio

----
The "Power" section heading is missing the necessary markup

----
Change:
>Located on the ICSP header only support

to:
>Located on the ICSP header **only. Support**

----
Change:
>When the pin is HIGH value,

to:
>When the pin is HIGH,

----
Change:
>By default they measure from ground to 3.3 volts, though is it possible 

to:
>By default they measure from ground to 3.3 volts, though it is possible 

----
Change:
>change the upper end of their range using the AREF pin and the analogReference() function.

to:
>change the upper end of their range using the AREF pin and the [analogReference()](https://www.arduino.cc/en/Reference/AnalogReference) function.

(Add link to the function reference page, consistent with the rest of the functions mentioned in this document.)

----
Change:
>AREF. Reference voltage for the analog inputs.

to:
>**AREF:** Reference voltage for the analog inputs.

----
Change:
>Reset. Bring this line LOW

to:
>**Reset:** Bring this line LOW

----
>Uploading sketches to the SAMD21 is different from the AVR microcontrollers found in other Arduino & Genuino boards. 

The layout of this text as it wraps around the image is awkward:
[image]

----
Change:
>and unlike EEPROM, and it must be erased in pages before writing.

to:
>and unlike EEPROM, it must be erased in pages before writing.

----
Inconsistent capitalization of "Programming Port" and "Native USB Port" throughout the document.

----
Change:
>when you connect the Arduino Zero through the programming USB connector;

to:
>when you connect the Arduino Zero through the Programming Port USB connector;

----
Change:
>when you connect the Arduino Zero through the native USB connector.

to:
>when you connect the Arduino Zero through the Native USB Port connector.

----
Change:
>Further the main UART (serial port) is located on the same pins (0 and 1).

to:
>**Further,** the main UART (serial port) is located on the same pins (0 and 1).

kengdahl, 2018-03-15:
The board is not based on Atmega32U4. it is based on ATSAMD21G18. I don't know how it reacts when opening the serial monitor. I will reasearch this.

per1234, 2018-03-15:
I realize the Zero isn't ATmega32U4 but the statement is talking about AVR boards, which include the ATmega32U4. Looking back I think I was being too picky even bringing that issue up.

I guess to be completely accurate you could say something like:
>Most Arduino and Genuino boards based on AVR microcontrollers get a reset and restart sketch execution each time the Serial Monitor of the Arduino Software (IDE) is opened.

But I really don't think it's a big deal if you want to leave it as is.

kengdahl, 2018-03-15:
Everything should be fixed now

closed 2018-03-15 by kengdahl