arduino/Arduino #6542: Please add Paperino PL_microEPD Arduino Library

RobPo, 2017-07-23:
Hi, 
please add the following libraries:
https://github.com/RobPo/Paperino/tree/master/library/Paperino_PL_microEPD_Arduino_Library
https://github.com/RobPo/Paperino/tree/master/library/Paperino_BO_BMA250E_Arduino_Library

Thanks,
Robert

per1234, 2017-07-23:
From https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#is-my-git-repo-ok

>Is my Git repo OK?
>Your repo is OK if its root folder contains the file library.properties and the rest of the library source code. If your library is stored in a subfolder, your repo is not OK and we can't include your library in the list.

So you will need to split your repository, which contains two libraries, in to two repositories and put each library in the root of its repository.

ccliche, 2017-07-23:
Hi,
What do the libraries relate to?

I was able to find the answer to the Compile and Upload issue.  The Arduino software has an issue with getting the correct JAVA files or having access.  I downloaded ECLIPSE for JAVA and let it revert to Arduino for the IDE.  There are other options.  But, I wanted to confirm the JAVA relationship.  BINGO....all worked beautifully.

Carl

From: Robert Poser 
Sent: Sunday, July 23, 2017 2:06 PM
To: arduino/Arduino 
Cc: Subscribed 
Subject: [arduino/Arduino] Please add Paperino PL_microEPD Arduino Library (#6542)

Hi,
please add the following libraries:
https://github.com/RobPo/Paperino/tree/master/library/Paperino_PL_microEPD_Arduino_Library
https://github.com/RobPo/Paperino/tree/master/library/Paperino_BO_BMA250E_Arduino_Library

Thanks,
Robert

per1234, 2017-07-23:
@ccliche that is completely irrelevant to the subject of this issue. If you're having trouble using Arduino please:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post the full contents of the error messages you're getting, using code tags (`</> button on the toolbar).

cmaglie, 2017-07-24:
@RobPo 
to add your libraries to the library manager you must create a separate github repository for each library (as pointed out by @per1234), and tag a release, please read carefully  https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Also your library.properties is missing some fields like `paragraph`, I suggest you to pick some already made library.properties file as an example.

For now I'm closing this one, please reopen when you're ready.

RobPo, 2017-07-24:
OK thanks for your feedback!
I now merged both libraries to one lib in one repository; being in the root of the repository:
https://github.com/RobPo/Paperino
The file library.properties is updated as suggested.
Please can you review adding my library again?

cmaglie, 2017-07-25:
Done, thanks!

closed 2017-07-25 by cmaglie

RobPo, 2017-08-02:
@cmaglie Please could you help: I understood the library can now be downloaded via the Library Manager of the Create Arduino Web IDE... is that correct? I am asking since I can't find it although there has been a sunday inbetween since closing this issue. Thanks in advance!

per1234, 2017-08-02:
It looks like no Library Manager libraries have been added to the Arduino Web Editor since [EasyNTPClient](https://github.com/arduino/Arduino/issues/6441). Everything from [OctoSonar ](https://github.com/arduino/Arduino/issues/6445)and newer does not show up in the Arduino Web Editor Library Manager search.

RobPo, 2017-08-07:
OK. Should we open a separate issue for this??

RobPo, 2017-08-08:
@per1234 Please could you give an indication whether this is a longer issue (hopefully not) or something that can be fixed in the next days? We have shipped the related hardware these days and if its not working the documentation would need to be changed on our side. Thanks in advance!

per1234, 2017-08-08:
@RobPo I'm not in a position to be able to give you any assurances on this matter. I'm just a volunteer who tries to help out with managing this issue tracker. I did create a dedicated issue report for the problem of the Arduino Web Editor's Library Manager not being updated because it's a separate problem from the original subject of this issue: https://github.com/arduino/Arduino/issues/6593