arduino/Arduino #6543: Were these malicious deletions in the wiki?

adamvoss, 2017-07-24:
I was trying to follow an external link to the wiki to the Boards Manager FAQ only to find the link dead.  It too a while but I found my information elsewhere.  I thought to perhaps recreate the page with a link to information I found helpful, but first I looked into history.

There I see commits c0f518e5bae7445ca4d4073f465f244a87f9a8d7 and a905bdd30a1bca6c7d7109ca6236c44cff709d6c both by @Cobus002 from 11 months ago.  @Cobus002 does not seem especially active on GitHub or seem to have any particular ties to the Arduino projects, which makes me wonder:  were these deletions approved of?

So you don't have to clone the wiki to get information on the commits here is the `git log --summary`:

```
commit a905bdd30a1bca6c7d7109ca6236c44cff709d6c
Author: Cobus002 <hoffmann.cobus@hotmail.com>
Date:   Sun Aug 23 18:25:02 2015 +1000

    Destroyed Boards Manager FAQ (markdown)

 delete mode 100644 Boards-Manager-FAQ.md

commit c0f518e5bae7445ca4d4073f465f244a87f9a8d7 (HEAD)
Author: Cobus002 <hoffmann.cobus@hotmail.com>
Date:   Sun Aug 23 18:24:53 2015 +1000

    Destroyed Git Guide for Arduino (markdown)

 delete mode 100644 Git-Guide-for-Arduino.md
```

matthijskooijman, 2017-07-24:
@Cobus002, could you comment on why you deleted these pages?

per1234, 2017-11-14:
@Cobus002, we're still waiting for a comment from you regarding these wiki deletions.

matthijskooijman, 2019-10-01:
For lack of response, I had a look at the pages themselves. I would say the board manager FAQ page is still useful and current, so I have restored it. The Git guide seems a bit detailed and specific, and mostly lists git commands that people should probably learn elsewhere, so I'm leaving that one out.

closed 2019-10-01 by matthijskooijman