arduino/Arduino #6548: Menu don't show up in OSX High Sierra Menubar

guidefreitas, 2017-07-25:
Menu don't show up in OSX High Sierra menubar.

S.O.: OSX High Sierra Beta 4

facchinm, 2017-07-25:
Thanks a lot for reporting! Could you debug the problem a bit while we setup a Mac with that OS version? Launching the IDE via commandline and pasting here the output would be enough. 
To launch the IDE simply execute `/Applications/Arduino.app/Contents/MacOS/Arduino` in a Terminal and any error should appear there. Thanks again!

cmaglie, 2017-07-25:
and here we go again... :-)

guidefreitas, 2017-07-25:
Hi. Thanks for the fast feedback. Here is the log.

Last login: Tue Jul 25 00:26:48 on ttys000
➜  ~ /Applications/Arduino.app/Contents/MacOS/Arduino
[JmDNS(fe80-0-0-0-aede-48ff-fe00-1122-en5.local.).Timer] WARN javax.jmdns.impl.tasks.resolver.DNSResolverTask - ServiceResolver(fe80-0-0-0-aede-48ff-fe00-1122-en5.local.).run() exception
java.io.IOException: No route to host (sendto failed)
	at java.net.PlainDatagramSocketImpl.send(Native Method)
	at java.net.DatagramSocket.send(DatagramSocket.java:693)
	at javax.jmdns.impl.JmDNSImpl.send(JmDNSImpl.java:1595)
	at javax.jmdns.impl.tasks.resolver.DNSResolverTask.run(DNSResolverTask.java:74)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
[fe80-0-0-0-aede-48ff-fe00-1122-en5.local..recover()] WARN javax.jmdns.impl.JmDNSImpl - RECOVERING
[JmDNS(fe80-0-0-0-aede-48ff-fe00-1122-en5.local.).State.Timer] WARN javax.jmdns.impl.tasks.state.DNSStateTask - Canceler(fe80-0-0-0-aede-48ff-fe00-1122-en5.local.).run() exception
java.io.IOException: No route to host (sendto failed)
	at java.net.PlainDatagramSocketImpl.send(Native Method)
	at java.net.DatagramSocket.send(DatagramSocket.java:693)
	at javax.jmdns.impl.JmDNSImpl.send(JmDNSImpl.java:1595)
	at javax.jmdns.impl.tasks.state.DNSStateTask.run(DNSStateTask.java:131)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
[fe80-0-0-0-aede-48ff-fe00-1122-en5.local..recover()] WARN javax.jmdns.impl.DNSStatefulObject$DefaultImplementation - Wait for canceled timed out: DNS: fe80-0-0-0-aede-48ff-fe00-1122-en5.local. [fe80:0:0:0:aede:48ff:fe00:1122%en5/fe80:0:0:0:aede:48ff:fe00:1122%en5] state: canceling 1 task: null
[fe80-0-0-0-aede-48ff-fe00-1122-en5.local..recover()] WARN javax.jmdns.impl.JmDNSImpl - fe80-0-0-0-aede-48ff-fe00-1122-en5.local.recover() Could not recover we are Down!

naturfenomen, 2017-07-26:
I'm on High Sierra public beta 3 and i's the same problem for me - i.e. no menu except for 'Arduino'. But no error when starting from Terminal.

facchinm, 2017-07-26:
Also reported [here](http://forum.imagej.net/t/fiji-or-imagej-lost-menu-in-new-macos-high-sierra/6222) and [here](http://help.infinitekind.com/discussions/problems/35345-md-loses-menu-bar-in-high-sierra-beta). It looks like this problem affects all Java8 programs, so it's very likely to be solved by Apple in one of the next betas. Keeping the issue open to track the issue anyway

cstatz, 2017-08-08:
This is still an issue for Public Beta 4!

jaxxon33, 2017-08-12:
Same issue with Developer Beta version 5, Aug 9 2017.

facchinm, 2017-08-14:
Argh, this is bad...

optikfluffel, 2017-08-14:
As a workaround for now you could install the [legacy Java 6 runtime](https://support.apple.com/kb/DL1572).

jaxxon33, 2017-08-19:
I have 2 MacBooks running High Sierra, one has Java 8 Update 121 and works fine with Arduino IDE, the other has Update 144 (currently latest) and has the 'no menu' problem.

vgiffin, 2017-08-26:
I saw this on Beta 3, but not on Beta 4 or later.

Surroot, 2017-09-08:
Still on developer beta 9 (17A360a) with Java 8 Update 144...

xDuk3x, 2017-09-21:
Same Error on Golden Master (17A362a) with Java 121 & 144

facchinm, 2017-09-21:
Adding other resources:
https://youtrack.jetbrains.com/issue/JRE-434#comment=27-2424694
https://bugs.eclipse.org/bugs/show_bug.cgi?id=520176

Trying to implement something similar to https://git.eclipse.org/r/#/c/105553/ and post a PR soon

facchinm, 2017-09-22:
Modifying `Info.plist` has no effect on OSX High Sierra GM . Following the discussion on Eclipse bugtracker, it looks like SWT has just been patched to overcome a bug in the latest OSX builds (`DefaultApp.nib` only present in `/System/Library/Frameworks/JavaVM.framework/Versions/A/Resources/en.lproj` folder)
see https://git.eclipse.org/r/#/c/105582/

to sum up:
- changing the WHOLE system locale to english works
- installing java6 doesn't work anymore on GM
- modifying `Info.plist` doesn't work in GM
- Arduino doesn't use SWT, but a similar workaround could be implemented in Swing (by Oracle/OpenJDK) sooner or later
- Apple could fix the bug before Monday 25/09 (release date)

In the meantime, let's use this thread to report any improvement.

McAlb, 2017-09-26:
I have just installed the brand spanking and official new High Sierra release. Menu bar is still not working. Changing the language to english works....

puybaret, 2017-09-26:
You could set `apple.laf.useScreenMenuBar` property to `true` only if the current system is not macOS 10.13 and the application menu isn't empty. Something like:

```
System.setProperty("apple.laf.useScreenMenuBar",
   String.valueOf(!System.getProperty("os.version").startsWith("10.13")
        || com.apple.eawt.Application.getApplication().isAboutMenuItemPresent()));
```

This is the solution I programmed in Sweet Home 3D. At least, the menu bar appears at the top of the window the time that Apple fixes the bug (just copy the `DefaultApp.nib` file of your locale from a previous macOS version and you'll see that the menu bar will go back to the screen menu bar, except if Apple decides to fix this issue another way).
Note that `isAboutMenuItemPresent` was removed under Java 9, where you should rather call `Desktop.isSupported(Desktop.Action.APP_ABOUT)` (you may also program that statement [with reflection](http://sweethome3d.cvs.sf.net/viewvc/sweethome3d/SweetHome3D/src/com/eteks/sweethome3d/MacOSXConfiguration.java?revision=1.61&view=markup#l100) if needed)

facchinm, 2017-09-26:
@puybaret , doesn't `apple.laf.useScreenMenuBar` need to be set before the GUI thread is spawned? Anyway, thanks for the tip, trying right now

puybaret, 2017-09-26:
It's not because you have to set this property before playing with Swing, that it cannot be set at the beginning of the program execution.

facchinm, 2017-09-26:
Here's the PR https://github.com/arduino/Arduino/pull/6763
I only tested it briefly on my colleague's OSX but I couldn't reproduce the issue (I should have rebooted probably, will try later). In the meantime, the build from the PR should be ready in 20 minutes, if anyone can test it it would be great.

puybaret, 2017-09-26:
Be sure to test on a system that doesn't use English as primary language

facchinm, 2017-09-26:
Sure, I changed the language to Italian but I didn't reboot, and the menus still appeared on a 10.13 final. Checking after a reboot later today. In the meantime, if you could test http://downloads.arduino.cc/javaide/pull_requests/arduino-PR-6763-BUILD-727-macosx.zip it would be supernice

McAlb, 2017-09-26:
I have just toggled back to german on PR-6763. Didn´t work... :-(

facchinm, 2017-09-26:
New build http://downloads.arduino.cc/javaide/pull_requests/arduino-PR-6763-BUILD-728-macosx.zip

felixerdy, 2017-09-26:
Build 728 still does not work on my system, language is german 😕

haciendalabs, 2017-09-26:
Hello, build 728 don't work here.french system.

facchinm, 2017-09-26:
Last surprise of the day, will continue tomorrow for the fix:
setting `apple.laf.useScreenMenuBar= false` in the build.txt has no effect either, but launching via CLI using system `java` and `CLASSPATH` brings the menus back in ANY language.
So tomorrow debugging session will be on Appbundler. 
Thanks everyone for testing, the [HELP WANTED] tag is still valid so if anyone can contribute with code and trials it would be supercool. Thanks again

Gohstgamer, 2017-09-26:
To all those who can't wait until an update fixes the bug, I have a work around:
1. Shutdown your Mac
2. Start in recovery mode and open the terminal then, type "csrutil disable"
3. Restart
4. Go to "/System/Library/Frameworks/JavaVM.framework/Versions/A/Resources/en.lproj"
5. Copy the file named "DefaultApp.nib"
6. Go to "/System/Library/Frameworks/JavaVM.framework/Versions/A/Resources/<your prefered language>.lproj"
7. Paste "DefaultApp.nib"
8. Move the "Menu.strings" from you prefered language folder to another folder (for eg. create an "Java <your language>.lproj" in your document folder), so you can restore all changes if you want to
9. Restart your System

Now Arduino should work again as it did, at least it worked for me as I placed the "DefaultApp.nib" into the "German.lproj" and deleted there "Menu.strings".

This worked for me, that doesn't mean it has to work for all of you or that's impossible something fails due to these changes. Do this at your own risk!

GaelC92, 2017-09-27:
Since it seems related to the language and country selection, it might help to use java switches 
java -Duser.language=en -Duser.country=US -Duser.variant=US -jar ...  to enforce locale settings. 
Unfortunately I'm unable to test, since I'm not sure which jar file / class should be called from CLI. 
Any hint ?

facchinm, 2017-09-27:
Hi @GaelC92 , thanks for the tip, unfortunately enforcing the locale via java switches doesn't solve the issue, probably due to AppBundler
Sample patch:
```patch
diff --git a/build/build.xml b/build/build.xml
index d95f8def6..b45dff0f7 100644
--- a/build/build.xml
+++ b/build/build.xml
@@ -411,6 +411,9 @@
       <option value="-Dcom.apple.smallTabs=true"/>
       <option value="-DAPP_DIR=$APP_ROOT/Contents/Java"/>
       <option value="-Djava.ext.dirs=$JVM_RUNTIME/Contents/Home/lib/ext/:$JVM_RUNTIME/Contents/Home/jre/lib/ext/"/>
+      <option value="-Duser.language=en"/>
+      <option value="-Duser.country=US"/>
+      <option value="-Duser.variant=US"/>
 
       <!--
       <option value="-Dapple.awt.showGrowBox=false"/>
```

GaelC92, 2017-09-27:
It may be necessary to pass LANG environnement vars as well... This is a workaround, but if somehow we manage to setup the JVM to use en_US locale, it should show the menu, according to the previous comments. 
So it's just a matter of how :)

GaelC92, 2017-09-27:
@facchinm could you provide me the main class name to launch the GUI ? (I'm struggling to find it within the code, since I'm new to it)

facchinm, 2017-09-27:
The main class is `Processing.app.Base`; I'm trying to set LANG from Info.plist right now, will report back as soon as I have any result.

GaelC92, 2017-09-27:
Got it working !! 
Here is how I managed to get the menus from CLI
1) export _JAVA_OPTIONS=" -Duser.language=en -Duser.country=US -Duser.variant=US -Duser.region=US "
(this enforces the java settings for each jvm / java tools and has priority over anything else. Alternatively one can use JAVA_TOOLS_OPTIONS, but this is lower priority than any other mean to set these values)
2) call the Base class this way (from the Java directory in the package) : 
LANG=en java -cp "arduino-core.jar:./*" processing.app.Base

GaelC92, 2017-09-27:
As a temporary in-code solution, you could maybe set the above variables at launch, conditioned on the JVM/OS version, and/or a preference setting ?

facchinm, 2017-09-27:
I just pushed another couple of patches on https://github.com/arduino/Arduino/pull/6763; newer appbundler allows setting environment variables via Info.plist, let's hope it's enough to workaround the issue

facchinm, 2017-09-27:
New build: http://downloads.arduino.cc/javaide/pull_requests/arduino-PR-6763-BUILD-729-macosx.zip

GaelC92, 2017-09-27:
nope. the new build does not fix the problem.

GaelC92, 2017-09-27:
Ok. Funny enough, the following command line *does* work, and displays the menu in french ! : _JAVA_OPTIONS="-Xdebug -Duser.language=fr -Duser.country=FR -Duser.variant=FR -Duser.region=FR " LANG=fr java -cp "arduino-core.jar:./*" processing.app.Base

GaelC92, 2017-09-27:
hmm... not sure what happened but now, I can simply call  java -cp "arduino-core.jar:./*" processing.app.Base" and it does work...

facchinm, 2017-09-27:
I've finally been able to find where the global menubar was being set forcefully; integrating it with @puybaret instructions gave the correct result (eg. the Windows-style workaround).
Build from @ArduinoBot is coming

facchinm, 2017-09-27:
Here we go again http://downloads.arduino.cc/javaide/pull_requests/arduino-PR-6763-BUILD-731-macosx.zip

facchinm, 2017-09-27:
[image]
Ugly but at least usable

McAlb, 2017-09-27:
yeah! You did it, it works. In my case in german... Thanks!!

GaelC92, 2017-09-27:
For what it's worth, here is what I obtain from the previous version when calling from command line. 
This hints that the problem probably does neither lie within JRE nor within the application code. 
I believe it's the appbundler but I cannot find what it does to make the menu fail.
@facchinm , maybe you could spot a difference in the env vars or java settings passed to the main class at startup ? (I could not until now)

[image]

facchinm, 2017-09-27:
I just merged the patch on `master` so it would be supercool if you could test the [hourly build](https://www.arduino.cc/en/Main/Software#hourly) as soon as it's ready. If everything is ok we can release soon and provide a workaround for all mac users willing to update. Thanks everyone for the help.

McAlb, 2017-09-27:
hourly build works like PR-6763-BUILD-731 - this time with german menus

McAlb, 2017-09-28:
right! it really seem to be a java-issue

AdrianBuza, 2017-09-29:
I found a simpler workaround. Force the language to English for the .app.
Open a Terminal and run:
`defaults write cc.arduino.Arduino AppleLanguages '(en)'`

manolivette, 2017-09-29:
La solution de AdrianBuza ok pour moi.

fschlag, 2017-09-30:
@AdrianBuza solution works for Arduino on a German Mac OS X 10.13 without using the nightlies/dailies. I am using version 1.8.4.

kuerb, 2017-10-02:
@AdrianBuza After installing the Teensyduino Add-on, these defaults seem to be ignored.

As another workaround I've used the freeware [Language Switcher](http://www.tj-hd.co.uk/en-gb/languageswitcher/) set to always open the Arduino app in English. I am not sure which flags the app changes, but it seems to be working.

facchinm, 2017-10-02:
Cross referencing the official blog post https://blog.arduino.cc/2017/10/02/arduino-ide-1-8-5-hotfix-for-macos-high-sierra-users/

Closing as fixed, thanks everyone!

closed 2017-10-02 by facchinm

PaulStoffregen, 2017-10-03:
> After installing the Teensyduino Add-on, these defaults seem to be ignored.

Please post a message about this Teensy-specific problem here.

https://forum.pjrc.com/threads/46745-Teensyduino-1-40-Beta-2

Let's not bother Arduino with issues from Teensy.

jandrioli, 2018-02-13:
This is happening on Linux too. I'm on Ubuntu 16.04 x64, Java openjdk version "1.8.0_151"
OpenJDK Runtime Environment (build 1.8.0_151-8u151-b12-0ubuntu0.16.04.2-b12)
OpenJDK 64-Bit Server VM (build 25.151-b12, mixed mode)

The source of the problem is a failing DNS request. If I start the Arduino IDE with access to the internet, and then switch access-points, it will lockup and we get this "we are down!" log. Also, subsequently launching the Arduino IDE without access to internet brings the same problem.

facchinm, 2018-02-13:
@jandrioli the missing menubar on Linux is likely to be a VERY different kind of bug. Feel free to open another issue with all the needed information (like screenshots). The "we are down" message is generated by mdns resolver and doesn't affect anything except the ability to perform network uploads

jandrioli, 2018-02-13:
Hi, the problem is not related to menubar (anyway the UI subsystem is
totally another animal from MacOS to Linux).
The issue here is actually related to java classes *jmDNS* not being able
to resolve something.
I can't reproduce now but the logs were quite verbose with this "not able
to resolve" message multiple times.
The last error message is "could not resolve. we are down!"
And the Arduino IDE is frozen after that.

facchinm, 2018-02-14:
@jandrioli in fact, this is why I asked you to open a different issue to track the jmdns problem :smile: