arduino/Arduino #6553: [Library manager] Please add Grove_Digital_Light_Sensor library

xyu6, 2017-07-26:
Hi

Please add our new Tic library to the library manager:https://github.com/Seeed-Studio/Grove_Digital_Light_Sensor 

We are Seeedstudio. We have plenty of libraries and want to add into your library manager. is there any special process we can follow to add all the libraries? 

Thanks!

per1234, 2017-07-26:
@xyu6 you need to create a library.properties file that's compliant with the Arduino Library Specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

After that, create a release or tag in the repository that matches the `version` value in your library.properties file.

For more information please see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

cmaglie, 2017-08-02:
> We are Seeedstudio. We have plenty of libraries and want to add into your library manager. is there any special process we can follow to add all the libraries?

@xyu6 
You can send a link to all the libraries in a single issue, please don't open an issue for each library.
BTW let's fix the problems for this library, so you have a "reference" for all the other libraries.

per1234, 2017-08-22:
Closing due to lack of response or action. @xyu6, if you ever get the library to meet the requirements for Library Manager index inclusion please comment here and I will reopen.

closed 2017-08-22 by per1234