arduino/Arduino #6560: avrdude config file ignored? Wrong programmer settings.

Testi89, 2017-07-27:
When I select avrispmkII as programmer in the IDE, I get the following output:

```
/home/piet/Downloads/arduino-1.8.3/hardware/tools/avr/bin/avrdude -C/home/piet/Downloads/arduino-1.8.3/hardware/tools/avr/etc/avrdude.conf -v -patmega328p -cstk500v2 -Pusb -e -Ulock:w:0x3F:m -Uefuse:w:0xFD:m -Uhfuse:w:0xDA:m -Ulfuse:w:0xFF:m 

avrdude: Version 6.3, compiled on Jan 17 2017 at 11:00:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/piet/Downloads/arduino-1.8.3/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/home/piet/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : stk500v2
         Setting bit clk period        : 10.0
avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)

avrdude done.  Thank you.

Fehler beim Brennen des Bootloaders.
```

The modified bit clk period is well recognized, but the settings for "Using Port" and "Programmer" are totally wrong.

Tested with Ubuntu-MATE 17.04, kernel 4.10.0-28

facchinm, 2017-07-27:
Hi @Testi89 ,
`avrispmkII` uses libusb directly (and not any serial emulation) so any "Port" configuration gets correctly ignored. 
Is the programmer being recognized (eg. can you see it when you type `lsusb`) ? If so, check `udev` permissions to make sure if you can access the device without being root.

Testi89, 2017-07-27:
Yes, I can access the m328 with avrdude via command line with -P /dev/ttyUSB0:

```
piet@piet-desktop:~$ avrdude -c avrispmkII -p m328 -P /dev/ttyUSB0

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.04s

avrdude: Device signature = 0x1e9514 (probably m328)

avrdude: safemode: Fuses OK (E:FD, H:D9, L:FF)

avrdude done.  Thank you.

piet@piet-desktop:~$ 
```

Testi89, 2017-07-27:
dmesg output:
```
[14652.657041] usb 3-2: new full-speed USB device number 3 using xhci_hcd
[14652.802240] usb 3-2: New USB device found, idVendor=0403, idProduct=6001
[14652.802243] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[14652.802244] usb 3-2: Product:  USB SERIAL CONVERTER
[14652.804972] ftdi_sio 3-2:1.0: FTDI USB Serial Device converter detected
[14652.806440] usb 3-2: FTDI USB Serial Device converter now attached to ttyUSB0
```

Testi89, 2017-07-27:
Maybe it is not working because I have an USB->Serial adapter between PC and avrisp? And this interferes with the direct libusb implementation?

facchinm, 2017-07-27:
Mmmmh, the AVRISPMKII is this little guy... Are you sure you are using it? BTW, the problem is absolutely the (indirect) USB connection 

[image]

Testi89, 2017-07-27:
I am using avrisp, but it will work with avrispmkii settings (normally).
I selected also the serial port as /dev/ttyUSB0 in the IDE.

Testi89, 2017-07-27:
What I do not correctly understand is why in the call of avrdude the parameter "-Pusb" is included.

closed 2017-07-27 by Testi89

facchinm, 2017-07-27:
If you are using avrisp you should indeed select `AVR ISP` from the "Programmers" menu, which correctly expands the "Port" field (see https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/programmers.txt#L6). Instead, when selecting the MKII, `-PUSB` is added (https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/programmers.txt#L13)