arduino/Arduino #6568: [Library Manager] anduinoWiFi library for anduinoWiFi shield

briancarbs, 2017-07-28:
https://github.com/andium/Anduino/tree/master/libraries/AnduinoWiFi

Thanks!

per1234, 2017-07-28:
From https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#is-my-git-repo-ok:
>Is my Git repo OK?
Your repo is OK if its root folder contains the file library.properties and the rest of the library source code. If your library is stored in a subfolder, your repo is not OK and we can't include your library in the list.

In order to be included in Library Manager you will need to move the library to its own repository, then create a tag or release that matches the `version` value in its library.properties file.

Please post a link to the new repository once this has been done.

The same applies to your other requests:
- https://github.com/arduino/Arduino/issues/6565
- https://github.com/arduino/Arduino/issues/6566
- https://github.com/arduino/Arduino/issues/6567

cmaglie, 2017-08-02:
> Please post a link to the new repository once this has been done.

please also post all of them in one issue it's not needed to open an issue for each library.

briancarbs, 2017-08-18:
Ah! Didn't realize each library needed their own specific repository. Will do!

per1234, 2017-10-03:
@briancarbs we can't proceed with adding your libraries to the Library Manager index until you have created compliant repositories for them. If you no longer want your libraries added to the index please let us know so we can close this issue.

briancarbs, 2017-10-04:
Thanks for the reminder! This got moved to the back burner for a moment, apologies for keeping the issue open for so long. I'll get this fixed by next week. Thanks!

cmaglie, 2018-01-15:
I'm closing this for now. When you're ready please comment further and it will be reopened.

closed 2018-01-15 by cmaglie