arduino/Arduino #6571: Arduino 1.8.2 under FreeBSD 11.0-RELEASE does not find EEPROM.h, SPI.h, etc.

derneue, 2017-07-29:
I am trying to compile the BareMinimum sketch. So far no problems. But after adding EEPROM.h to the sketch I got this error:

`
/usr/local/arduino/arduino-builder -dump-prefs -logger=machine -hardware /usr/local/arduino/hardware -tools /usr/local/arduino/tools-builder -built-in-libraries /usr/local/arduino/libraries -libraries /home/tobias/Ard/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10802 -build-path /tmp/arduino_build_394750 -warnings=none -build-cache /tmp/arduino_cache_532435 -prefs=build.warn_data_percentage=75 -verbose /home/tobias/Ard/BareMinimum/BareMinimum.ino
/usr/local/arduino/arduino-builder -compile -logger=machine -hardware /usr/local/arduino/hardware -tools /usr/local/arduino/tools-builder -built-in-libraries /usr/local/arduino/libraries -libraries /home/tobias/Ard/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10802 -build-path /tmp/arduino_build_394750 -warnings=none -build-cache /tmp/arduino_cache_532435 -prefs=build.warn_data_percentage=75 -verbose /home/tobias/Ard/BareMinimum/BareMinimum.ino
Using board 'mega' from platform in folder: /usr/local/arduino/hardware/arduino/avr
Using core 'arduino' from platform in folder: /usr/local/arduino/hardware/arduino/avr
Detecting libraries used...
"/usr/local/arduino/tools-builder/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/usr/local/arduino/hardware/arduino/avr/cores/arduino" "-I/usr/local/arduino/hardware/arduino/avr/variants/mega" "/tmp/arduino_build_394750/sketch/BareMinimum.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/usr/local/arduino/tools-builder/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/usr/local/arduino/hardware/arduino/avr/cores/arduino" "-I/usr/local/arduino/hardware/arduino/avr/variants/mega" "/tmp/arduino_build_394750/sketch/BareMinimum.ino.cpp" -o "/tmp/arduino_build_394750/preproc/ctags_target_for_gcc_minus_e.cpp"
/home/tobias/Ard/BareMinimum/BareMinimum.ino:1:20: schwerwiegender Fehler: EEPROM.h: No such file or directory
Kompilierung beendet.
exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Mega or Mega 2560.
`
The same behaviour is shown for SPI.h. 

The Arduino Software is installed under 
`/usr/local/arduino`

The contents of arduino-builder.options:
`
{
  "hardwareFolders": "/usr/local/arduino/hardware",
  "otherLibrariesFolders": "/usr/local/arduino/libraries",
  "toolsFolders": "/usr/local/arduino/tools-builder",
  "runtime.ide.version": "10612"
}
`
Has anybody the same issues?

per1234, 2017-07-29:
From this "arduino-builder.options" thing, it [looks ](https://lists.freebsd.org/pipermail/svn-ports-all/2016-October/134680.html)like the FreeBSD package maintainers have mucked around with the Arduino IDE. Please install the official Arduino IDE 1.8.3 IDE downloaded from:
https://www.arduino.cc/en/Main/Software
Let me know if that solves the problem.

per1234, 2017-08-14:
@derneue did you ever try the official version of the Arduino IDE? If so, did it solve this issue?

derneue, 2017-08-14:
I am in contact with the port maintainer to resolve this issue.

per1234, 2017-08-14:
OK, I'm going to interpret that as saying that the problem was caused by the use of a modified IDE and thus closing this issue as invalid.

closed 2017-08-14 by per1234