arduino/Arduino #6575: Missing circuit wiring in "TFT Bitmap Logo"

ad05bzag, 2017-08-01:
https://www.arduino.cc/en/Tutorial/TFTBitmapLogo tutorial is missing two connections:
MISO: pin 12
SD CS: pin 4
as per http://www.arduino.org/learning/getting-started/getting-started-with-arduino-robot-lcd
without these it was not possible to complete the tutorial. It would halt at: 
"Initializing SD card...Failed"

SimonePDA, 2017-12-27:
The SD card requires MISO, MOSI, Clock and ChipSelect (CS) pins connected to be available. The Example has the proper #define sd_cs  4 line in the code and uses SPI lib for communications. By default Arduino UNO uses for SPI the following MOSI - pin 11, MISO - pin 12 and CLK - pin 13. CS is defined in the code. 
The fritzing scheme is correct, but the textual explanation is missing CD and MISO connections.
I'm fixing it.

SimonePDA, 2017-12-27:
Pushed online

per1234, 2018-01-15:
@SimonePDA the large size image (https://www.arduino.cc/en/uploads/Tutorial/TFT_logo_large.png) is fixed:
[image]
but the small size image shown on the tutorial page (https://www.arduino.cc/en/uploads/Tutorial/GTFT_logo3.png) is not fixed:
[image]

SimonePDA, 2018-03-17:
Fixed

closed 2018-03-17 by SimonePDA