arduino/Arduino #6579: SPI examples lack begin/endTransaction()

septillion-git, 2017-08-02:
Like the title says, the SPI examples in the IDE don't show how to use beginTransaction(), endTransaction() or SPISettings. Think it would be useful to add it to them.

septillion-git, 2018-04-02:
Then use the default values! It's not as if the default value is a magic value that's not peripheral dependent. People tend to just NOT use them and only use the old legacy way of applying settings with SPI.setBitORder(), SPI.setClockDivider() and SPI.setDataMode().

I think the examples should show as much of the functionality the library has to offer and how to use it. Because it's not in an example people don't use it and it will break the whole idea of being able to run the SPI at different speeds in a sketch. Especially if the magic of a library is used. The default speed isn't restored after a library correctly uses beginTransmission() and endTransmission() aka will result in unknown speed changes for newbies.

I just think it's weird that if you say "because the values of the function should be adjusted according to the data found on the peripheral's datasheet that you want to connect" you just ignore setting the speed at all for the examples. 😩 The examples are written for specific peripherals, the SCP1000 and the AD5206.

PaulStoffregen, 2018-04-02:
Agreed, the examples should be updated.  @SimonePDA - Please reopen this issue.

The old way does not even give consistent results on Arduino's newer products, like MKR1000, Arduino Zero or Arduino Due.

> The examples provided for SPI are made with components that implement the most common settings and therefore work with the default parameters. You can find an explanation of these parameters on the SPI Library reference page. (https://www.arduino.cc/en/Reference/SPI)

If you actually read that page, it clearly says to use SPI.beginTransaction().

> Once you have your SPI parameters, use SPI.beginTransaction() to begin using the SPI port. The SPI port will be configured with your all of your settings. The simplest and most efficient way to use SPISettings is directly inside SPI.beginTransaction()  ....

Also mentioned on that page is the fact that beginTransaction() protects your access to SPI with other libraries with interrupts attempt to access SPI.  Even though the examples do not use interrupts, Arduino's examples should demonstrate proper technique.  Novice users often copy and paste example code into larger projects using libraries.  They do not know how the libraries are designed.  They do not know whether a specific library uses interrupts.  Arduino's published examples should be written with the recommended good practices, as explained on that page.

septillion-git, 2018-04-02:
Exactly! 👍 That's indeed my point.