arduino/Arduino #6585: avr-g++: error: missing device or architecture after '-mmcu='

ChillarAnand, 2017-08-04:
When running arduino from rasp pi cli, it is not passing mmcu.

```
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
/home/pi/projects/vendor/arduino/arduino-builder -dump-prefs -logger=machine -hardware /home/pi/projects/vendor/arduino/hardware -tools /home/pi/projects/vendor/arduino/tools-builder -tools /home/pi/projects/vendor/arduino/hardware/tools/avr -built-in-libraries /home/pi/projects/vendor/arduino/libraries -libraries /root/Arduino/libraries -fqbn=arduino:avr:mega -vid-pid=0X2341_0X0042 -ide-version=10804 -build-path /tmp/arduino_build_293002 -warnings=null -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/pi/projects/vendor/arduino/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/home/pi/projects/vendor/arduino/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/home/pi/projects/vendor/arduino/hardware/tools/avr -verbose /tmp/arduino.ino
/home/pi/projects/vendor/arduino/arduino-builder -compile -logger=machine -hardware /home/pi/projects/vendor/arduino/hardware -tools /home/pi/projects/vendor/arduino/tools-builder -tools /home/pi/projects/vendor/arduino/hardware/tools/avr -built-in-libraries /home/pi/projects/vendor/arduino/libraries -libraries /root/Arduino/libraries -fqbn=arduino:avr:mega -vid-pid=0X2341_0X0042 -ide-version=10804 -build-path /tmp/arduino_build_293002 -warnings=null -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/pi/projects/vendor/arduino/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/home/pi/projects/vendor/arduino/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/home/pi/projects/vendor/arduino/hardware/tools/avr -verbose /tmp/arduino.ino
Using board 'mega' from platform in folder: /home/pi/projects/vendor/arduino/hardware/arduino/avr
Using core 'arduino' from platform in folder: /home/pi/projects/vendor/arduino/hardware/arduino/avr
Detecting libraries used...
"/home/pi/projects/vendor/arduino/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu= -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/pi/projects/vendor/arduino/hardware/arduino/avr/cores/arduino" "-I/home/pi/projects/vendor/arduino/hardware/arduino/avr/variants/mega" "/tmp/arduino_build_293002/sketch/arduino.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/home/pi/projects/vendor/arduino/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu= -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/pi/projects/vendor/arduino/hardware/arduino/avr/cores/arduino" "-I/home/pi/projects/vendor/arduino/hardware/arduino/avr/variants/mega" "/tmp/arduino_build_293002/sketch/arduino.ino.cpp" -o "/tmp/arduino_build_293002/preproc/ctags_target_for_gcc_minus_e.cpp"
avr-g++: error: missing device or architecture after '-mmcu='
```

Here is the same working correctly from ubuntu cli

```
"/home/chillaranand/projects/vendor/arduino/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/chillaranand/projects/vendor/arduino/hardware/arduino/avr/cores/arduino" "-I/home/chillaranand/projects/vendor/arduino/hardware/arduino/avr/variants/mega" "/home/chillaranand/projects/vendor/arduino/hardware/arduino/avr/cores/arduino/WMath.cpp" -o "/tmp/arduino_build_178850/core/WMath.cpp.o"
"/home/chillaranand/projects/vendor/arduino/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections
 -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/chillaranand/projects/vendor/arduino/hardware/arduino/avr/cores/arduino" "-I/home/chillaranand/projects/vendor/arduino/hardware/arduino/avr/variants/mega" "/home/chillaranand/projects/vendor/arduino/hardware/arduino/avr/cores/arduino/WString.cpp" -o "/tmp/arduino_build_178850/core/WString.cpp.o"

```

matthijskooijman, 2017-08-04:
You didn't say how you are calling the Arduino IDE on the CLI, which seems to be essential information here. The builder commandline contains `-fqbn=arduino:avr:mega` which suggests that is the board name you are passing on the commandline. However, it is missing the `cpu` option that distinguishes between the Mega with 2560 and 1280.

When I select the mega2560 in the Arduino IDE GUI and run verify, I see `-fqbn=arduino:avr:mega:cpu=atmega2560` in the verbose output, which I think should fix your issue.

ChillarAnand, 2017-08-04:
@matthijskooijman I am using this command 
```
arduino --port /dev/ttyACM* --board arduino:avr:mega --upload foo.ino
```

per1234, 2017-08-04:
Try this:
```
arduino --port /dev/ttyACM* --board arduino:avr:mega:cpu=atmega2560 --upload foo.ino
```

ChillarAnand, 2017-08-04:
Thanks. It fixed the issue. Is it documented anywhere?

closed 2017-08-04 by ChillarAnand

facchinm, 2017-08-04:
https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc#options :wink:

matthijskooijman, 2017-08-04:
@facchinm, perhaps it would make sense for the IDE or arduino-builder to detect a missing parameter and error out (listing all possible values for the parameter perhaps)? AFAICS it is never valid to omit a parameter, right (the GUI doesn't allow this either)? This is a common problem it seems, and listing the valid fqbns would be helpful to users.

facchinm, 2017-08-04:
I agree about adding a command to list available fqbns. Don't have any strong opinion about failing on missing parameter, though.