arduino/Arduino #6590: Better documentation of open hardware licenses and file availablity

adamvoss, 2017-08-07:
The [FAQ](https://www.arduino.cc/en/Main/FAQ#toc3) indicates the design files are licensed under a CC-BY-SA license.  However, when I go to [a product page](https://store.arduino.cc/usa/arduino-due) I can download the schematic files but there is no indication of license or license terms there, nor have I been able to find anywhere saying which ***specific*** license the files are under and I am left not knowing which license I would need to adhere to.

per1234, 2017-08-07:
If you look at the list of endorsements at https://www.oshwa.org/definition/ you will notice that Massimo Banzi, David A. Mellis, David Cuartielles, and Tom Igoe are all on the list.

If Arduino is going to sell products advertised as "open-source hardware!" it only makes sense that they should meet the open source hardware definition they have endorsed.

In the recent blog post: [A new era for Arduino begins today](https://blog.arduino.cc/2017/07/28/a-new-era-for-arduino-begins-today/) Mr Banzi said:
>This is the beginning of a new era for Arduino in which we will strengthen and renew our commitment to open source hardware and software

Surely the first step towards that would be a comprehensive review of all license documentation and fixing any issues that are found.

q2dg, 2017-08-08:
...and, of course, clean/clarify the .org stock