arduino/Arduino #6598: Include all file in directory!

MagoKimbra, 2017-08-09:
Hi, sorry my bad English.
I have this need, in my scretch I include the files in src/test 1.h, 2.h, 3.h.
#include "src/test/1.h"
#include "src/test/2.h"
#include "src/test/3.h"

If I wake up the file 4.h you will also need to add the corresponding include.
There is a way to include the entire folder type:
#include "src/test/* .h"
Or
#include "src/test"

Of course, these two commands have tried them but they do not go on. Since I often add commands to the test folder it would be nice to be able to do it.

Thank you!

facchinm, 2017-08-10:
Hi @MagoKimbra ,
you usually need to add an header file if you want to add new functions to your program, so using a wildcard for includes has never been considered for C/C++.  

Adding it by ourselves (as a preprocessing step) is feasible but there are very little advantages in doing so; unlike Java, the namespace you are going to use to call the functions included in a "package" doesn't depend on the file you included.

However, if you want automatic prototype adder, we have it :smile: ; if a function is declared inside a file with `.ino` extension in the same folder of the main sketch, its prototypes will be automatically added and you can use the function throughout any project file.

closed 2017-08-10 by facchinm

MagoKimbra, 2017-08-10:
Thank you for the reply...
If you can look [here](https://github.com/MKFirmware/MK4duo/tree/master/MK4duo) I have a set of commands inside the folder src/gcode/<category>/file.h
Virtually to add a command is enough to create the file corresponding to the command, but then I have to include it otherwise it does not go. I would have liked to have a solution that would implement the command automatically when the file was created.
The call to the function is already automatic because it creates a void table that fills only if the command is defined, in fact, to start file put #define CODE_<nome command>.
I did not understand the file .ino!!
Thanks anyway...