arduino/Arduino #6600: Arduino CC 1.8.3 on dual monitor does not work

Clicker79, 2017-08-09:
Arduino CC 1.8.3

I have 2 monitors and when I launch Arduino CC and move one to my second monitor the CC editor becomes all fuzzy and cannot see anything.  Works fine on my Primary monitor.

Primary monitor is 1920x1200
Second monitor is 1680x1050

I was able to to use dual monitors with single or multiple instance of CC with 1.8.1 version.

facchinm, 2017-08-10:
Hi @Clicker79 ,
could you provide a screencast of what happens? I believe it could be due to a conflict between your graphic card drivers and the Java runtime we bundle (something like https://github.com/arduino/Arduino/issues/5918).
Could you provide your PC specifications (OS version, processor, graphic card)?
Also, we updated the Java runtime in hourly build (download it from [here](https://www.arduino.cc/en/Main/Software)); if you could test it too it would be great.

Clicker79, 2017-08-10:
See attached file Capture.JPG taken from the 1.8.4 hourly build.  The same 2 lines are repeating when I scrolled down in the editor.

Same thing happens when I use version 1.8.3.  Also, scroll bar up/down buttons don't work.

I'm working on Window 7 Enterprise 64-bit, Intel i7-4810MQ, 16GB RAM, AMD FirePro W4170M.

[image]

Zerick, 2017-08-31:
I have the same or a similar problem, my display gets mangled badly on 3 out of 4 monitors, on the 4th is remains mangled, but is able to self refresh on a resize.

This only happens with the Arduino IDE, I've upgraded to the latest 1.8.4

[image]

cmaglie, 2017-08-31:
@Zerick can you post your graphics card and PC hardware model?

Zerick, 2017-08-31:
[image]

[image]

cmaglie, 2017-08-31:
@Zerick @Clicker79 
You're experiencing the same problem and you both have the "AMD FirePro 41xx" graphics card.
This makes me think that it's a driver problem, may you check from the producer website if there is an update to the graphics driver for your video card?

Zerick, 2017-08-31:
that *IS* interesting, my previous video card had the same problem and it was a FirePro V5900

Clicker79, 2017-09-11:
I updated the drivers and problem still exists. I tried the latest hourly build also. All my other applications are and have been working fine with AMD FirePro.  There is a bug in the Arduino CC.  I'm about ready to drop this product for the next design respin.

Zerick, 2017-09-11:
I've just switched to a Quadro P1000 and the Arduino IDE still only works on one of four monitors.  Video artifacting on the other 3 still occurs though it isn't quite as bad.

facchinm, 2017-09-12:
@Zerick @Clicker79 you could try adding a couple of lines in `arduino.l4j.ini` (in Arduino installation root)
```
-Dsun.java2d.d3d=false
-Dsun.java2d.noddraw=true
```

This could solve most swing-related Java bugs
source: https://stackoverflow.com/a/3604200

alangbeen, 2017-09-12:
In response to issue #6636, Eclipse runs fine on all three of my monitors, and 1 is rotated.  Arduino still has issue on the rotated display. 
 I need to log in as admin to try the suggestion above.  Will do that later today.

Zerick, 2017-09-12:
@facchinm Thanks for the suggestion.  I'm sad to report, I made the suggested changes above and I've seen no change in behavior.
I also tried removing all java from my computer (windows 7 desktop) and reinstalling the latest, no effect with or without your suggested changes.

alangbeen, 2017-09-12:
I have also tested this with no improvements in behavior.  It appears to be a problem with the window redraw:  anything I type when the window is placed on the affected display shows up when I move the window back to the primary monitor.  I should also note that the window does not necessarily do a full refresh when I move it back and a minimize/restore action is necessary to see all features again in some instances.

facchinm, 2017-09-12:
Thanks you so much for testing, one last request:
if you start Arduino from `arduino_debug.exe` (in the same folder of the main executable) is Java reporting any exception? If so, could you paste them here? Thanks again

alangbeen, 2017-09-12:
I did some more searching and found [**this blog post**](https://www.neunbeere.de/blog/2017/08/graphics-bug-in-windows-7-after-installing-august-2017-security-updates/) on the topic (linked from [**this**](https://social.technet.microsoft.com/Forums/windowsserver/en-US/2c0ea396-bec2-4d8e-972f-13f19fc35f9f/multiple-monitors-office-not-refreshing-after-update-installed?forum=winserverTS) forum thread).  It appears to be a problem with the Windows operating system and hinges around a couple security patches.
The author list some work-arounds.  Since I'm on a work PC, I not going to be allowed to remove any security patches, but I did test the monitor orientation workaround and found that everything works correctly when I position the primary monitor farthest to the left.
The author lists some work-arounds for the issue.  Since I'm on a work PC, I not going to be allowed to remove any security patches, but I did test workaround for monitor orientation and found that everything works correctly when I position the primary monitor farthest to the left.

Zerick, 2017-09-12:
I haven't fiddled with the patch yet, but making my TOP LEFT screen primary solved the problems I was having with the Arduino IDE AND with Sketchup.
@alangbeen Nice find!

facchinm, 2017-09-13:
@alangbeen super nice find! I'm only bugged by Eclipse working correctly in the same setup... I'll investigate a bit more

justinmclean, 2017-09-13:
Crewf.g BV CFE
By be new g breed

alangbeen, 2017-09-13:
@facchinm 
[image]