arduino/Arduino #6602: Please add Charge-n-Boost library to library manager

bernebeer, 2017-08-10:
Hi,

Please add my library: Charge-n-Boost.

https://github.com/bernebeer/Charge-n-Boost/releases/tag/1.1

Thanks!

Gijs

bernebeer, 2017-08-10:
Was not Semversioning compliant. Changed it to 1.1.0

Link to release: https://github.com/bernebeer/Charge-n-Boost/releases/tag/1.1.0

cmaglie, 2017-08-14:
@bernebeer 
the library name `Charge 'n Boost` cannot be used due to the `'` on the name, may you remove the quote?
(remember to change the name in the `library.properties` and to commit and push **before** redoing the tag)

bernebeer, 2017-08-14:
Hi cmaglie,

Thank you for getting back to me. I have updated the library to reflect the new name.: 

https://github.com/bernebeer/Charge-n-Boost/releases/tag/1.1.0

Regards,

Gijs

cmaglie, 2017-08-14:
Library added, thanks!

closed 2017-08-14 by cmaglie