arduino/Arduino #6607: Library documentation

z3t0, 2017-08-13:
I would like to add information to the site [here](https://www.arduino.cc/en/Tutorial/LibraryExamples).

per1234, 2017-08-13:
The best way to suggest changes to that page is to submit an issue report to this repository. It's helpful if you can specify the exact wording of the change you would like rather than just a vague description of the suggested improvement.

z3t0, 2017-08-13:
I am currently the maintainer of the Arduino-Irremote library. Currently I am spending time updating documentation and writing tutorials for the library. I think it would be a great idea to add the tutorials to the site I mentioned above.

per1234, 2017-08-13:
Sorry, that page is for official Arduino libraries only so that will not be possible. I would recommend you to instead use the GitHub wiki feature in your repository for this purpose.

closed 2017-08-13 by per1234

z3t0, 2017-08-13:
Ah sorry. I misunderstood. Thanks!

Thanks,
Rafi Khan

Closed #6607<https://github.com/arduino/Arduino/issues/6607>.