arduino/Arduino #6609: error: deducing from brace-enclosed initializer list requires #include <initializer_list>

Sod-Almighty, 2017-08-13:
One of the things that really gets up my nose is when the compiler _explicitly instructs_ the programmer to insert code that blatantly does not work.

```
auto CHIP_MAPPINGS = {PC3, PC2, PC1, PC0, PB2, P1, P2, P3, PB1, P4, P5, P6, P7, P0, PB0, PD7};
```
> error: deducing from brace-enclosed initializer list requires #include <initializer_list>

Okay, so l do what I'm told:

```
#include <initializer_list>
auto CHIP_MAPPINGS = {PC3, PC2, PC1, PC0, PB2, P1, P2, P3, PB1, P4, P5, P6, P7, P0, PB0, PD7};
```
> fatal error: initializer_list: No such file or directory

Might I suggest that you modify your compiler so that it doesn't blatantly lie to its users?

facchinm, 2017-08-14:
Hi @Sod-Almighty ,
which board are you targeting? It doesn't look like a standard Arduino (or, at least, somechips define similar symbols but other are missing).
You should contact the core "vendor" and report the bug, which is probably due to a missing 
```-std=c++11``` in the compilation command line.
Closing as wontfix, feel free to reopen if you think that the problem applies to an arduino provided core, thanks.

closed 2017-08-14 by facchinm

Sod-Almighty, 2017-08-14:
I'm using a bare ATmega328 on a breadboard, with this `boards.txt` definition:
```
atmega328bb.name=ATmega328 on a breadboard (8 MHz internal clock)

atmega328bb.upload.protocol=arduino
atmega328bb.upload.maximum_size=30720
atmega328bb.upload.speed=57600

atmega328bb.bootloader.low_fuses=0xE2
atmega328bb.bootloader.high_fuses=0xDA
atmega328bb.bootloader.extended_fuses=0xFD

atmega328bb.bootloader.file=atmega/ATmegaBOOT_168_atmega328_pro_8MHz.hex
atmega328bb.bootloader.unlock_bits=0x3F
atmega328bb.bootloader.lock_bits=0x0F

atmega328bb.build.mcu=atmega328p
atmega328bb.build.f_cpu=8000000L
atmega328bb.build.core=arduino:arduino
atmega328bb.build.variant=arduino:standard
atmega328bb.build.board=AVR_ATMEGA328BB

atmega328bb.bootloader.tool=arduino:avrdude
atmega328bb.upload.tool=arduino:avrdude
```
Looks like standard Arduino core to me.

facchinm, 2017-08-14:
I misspoke, sorry. I intended that you are not targeting an official Arduino board (or core) (which in fact you are not doing, since we don't provide the board files for `328 on breadbord` "variant".
Also, if I try to compile your sample code targeting an Arduino Diecimila (which shares the processor you are targeting) I get errors like
```
'P1' was not declared in this scope
```

This is more appropriate topic to discuss on the Arduino Forum.

Please do this:

Read http://forum.arduino.cc/index.php?topic=148850
Create a new thread in the appropriate thread of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</> button on the toolbar).
Add a comment to this issue with the link to your thread to help anyone else who later has the same problem find the solution.

facchinm, 2017-08-14:
After removing all the missing defines, however, I'm able to reproduce the bug.
It looks like avr-gcc provided by Atmel doesn't include `initializers_list` header, and an unofficial repo has been created to fix this https://github.com/the2masters/avr-initializer_list.
I agree that the error message from the compiler is not great at all, I'll take care pinging the mainline maintainer to get the hint only available if the header is in fact there.

Thanks

matthijskooijman, 2017-08-18:
The problem is that `initializer_list` is supposed to be part of libstdc++, which gcc assumes to be available, but is not available for AVR. There are some Arduino libraries are available that add uClibc++ to Arduino, which has that header. I haven't tested it, but I think this one is the recommended one to use right now: https://github.com/mike-matera/ArduinoSTL

stalek71, 2018-11-30:
Probably you would like to do something like this (requires c++17 option).
This code calculates values during compilation.
`template<typename ...What>
constexpr auto calc(What ...what)
{
	return (... + what);
}

template<typename T>
constexpr auto calcMask(T arg0)
{
	return 1 << arg0;
}

template<typename T, typename  ...TRest>
constexpr auto calcMask(T arg0, TRest ... args)
{
	return 1 << arg0 | calcMask(args...);
}

int main()
{
	auto result = calc(20 + 10 + 5 + 5);
	auto mask = calcMask(2, 4, 6);

	std::cout << "calc() value is " << result << std::endl;
	std::cout << "calcMask() value is " << mask << std::endl;
}
`
I took newer version of avr compiler from this website:
http://blog.zakkemble.net/avr-gcc-builds/