arduino/Arduino #6610: [Library Manager] MicroC/OS Arduino library

Gibartes, 2017-08-14:
MicroC/OS for arduino which based on ARM Cortex-M3 like Arduino Due.
https://github.com/Gibartes/uCOS-II-Arduino.git

per1234, 2017-08-14:
From https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ:
>Is my Git repo OK?
Your repo is OK if its root folder contains the file library.properties and the rest of the library source code. If your library is stored in a subfolder, your repo is not OK and we can't include your library in the list.

Since your repository contains two libraries in subfolders it can not be added to Library Manager. Please either create a repository for each library with the library in the root or combine the two libraries into one and place in the root of the current repository and then comment here when your library meets all the requirements listed at the link above so we can proceed with adding it to Library Manager.

Gibartes, 2017-08-14:
Make a new repository : https://github.com/Gibartes/MicroC-OS-Arduino.git

per1234, 2017-08-14:
The library is still in a subfolder of the repository. You need to move it to the root of the repository.

After doing that you need to create a tag or release that matches the version value in your library.properties.

After that please comment here to let us know that your library meets all the requirements.

Gibartes, 2017-08-14:
Thanks for your advice. I attached tags and remove subfolders
repo : https://github.com/Gibartes/uCOS-II_Arduino

per1234, 2017-08-14:
@Gibartes why did you close this issue? Do you no longer want the library added to Library Manager?

cmaglie, 2017-08-14:
I've added the library to the indexer, but there is a problem, you must fill `author` and `maintainer` fields and also `paragraph` (with a sentence explaining the purpose of the library).

```
JOB 1059 - 2017/08/14 11:19:05 Scraping https://github.com/Gibartes/uCOS-II_Arduino.git
JOB 1059 - 2017/08/14 11:19:08 Checking out tag: 2.92.10
JOB 1059 - 2017/08/14 11:19:08 Error syncing library: Error generating library from repo: 'author' and 'maintainer' fields must be defined
```

Please:
1) update the library.properties
2) commit & push
3) redo the tag

after doing the above, the library will be automatically published within an hour (no need to further comment on this issue).

closed 2017-08-14 by cmaglie