arduino/Arduino #6611: Document library name restrictions in Library Manager FAQ

per1234, 2017-08-14:
I know there are some restrictions on the allowed library.properties `name` values in order for a library to be allowed inclusion in the Library Manager index. It seems like it would be a good idea to document this in the FAQ. I'm happy to do the edit but I don't know exactly what the exact restrictions are so I just need a little information to get started.

@cmaglie I guess you're the expert on this?

cmaglie, 2017-08-17:
> I'm happy to do the edit but I don't know exactly what the exact restrictions are so I just need a little information to get started

thank you @per1234!

The allowed chars for the library name are:

- numbers `0123456789`
- letters (any case) `A-Z` or `a-z`
- space, underscore, dot and dash ` _.-`

the name cannot start or end with a space.

cmaglie, 2017-08-17:
And it cannot start with a number too... see #6630... (sigh)

Even if this limitation is going to be removed (https://github.com/arduino/Arduino/issues/6592) it's better to keep it in the library manager to allow compatibility with older IDE installations.

KuangLei, 2017-08-17:
@cmaglie If the library name allows spaces, please also update the notification here.  
[image]

Thanks.

KuangLei, 2017-08-17:
@cmaglie @per1234 I've [update](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification/_compare/7126f14f9a8d3c832b61921d72edc1e82f0adc60) the [Arduino IDE 1.5: Library specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification) with following line.  
> **name** - the name of the library. Library names must contain only basic letters (`A`-`Z` or `a`-`z`) and numbers (`0`-`9`), spaces (` `), underscores (`_`), dots (`.`) and dashes (`-`). It cannot start or end with a space, and also it cannot start with a number.      

BTW, I think now the folder name of a library is not decided by the `name` in `library.properties`, because when I changed the `name` in `library.properties` of my library two hours ago, the problem didn't solved. So I hope the library server can distinguish library name automatically by the `name` in `library.properties`.  

Thanks.

cmaglie, 2017-08-17:
> BTW, I think now the folder name of a library is not decided by the name in library.properties

actually the problem is in the check for the folder name performed by the IDE that is made independently from the library manager.
I think that you still have a library **folder** named `4Digit7Seg12Pin` that is causing the popup to appear. BTW now that the name is fixed in the library manager this should not happen anymore.

KuangLei, 2017-08-17:
Thanks a lot, my library works well now.  
Two hours ago, the `name` in `library.properties` had been changed to `fDigitsSegtPin` (4Digit7Seg12Pin Release v1.0.2). But when I remove the `4Digit7Seg12Pin` folder in my `library` folder and re-download it though library manager, it create a `4Digit7Seg12Pin` folder again. So the IDE maybe create the folder with the name of the library's name on the library server but not with `name` in `library.properties`.

cmaglie, 2017-08-17:
> But when I remove the 4Digit7Seg12Pin folder in my library folder and re-download it though library manager, it create a 4Digit7Seg12Pin folder again

That was because the name cannot be change once the library is registered in the index (neither changing it from library.properties). Probably you tried before I changed the name in #6630

KuangLei, 2017-08-17:
Yes, I've tried to change the name before #6630 , so I hope that the server can automatically check and change the name in the index with `name` in `library.properties` when updating to a new version (maybe `sentence` and `paragraph` too; I didn't have a try). Thanks.

closed 2017-08-18 by per1234

feikname, 2017-08-19:
@cmaglie It might a good idea to not allow major OS reserved names too.

* [Windows](https://msdn.microsoft.com/en-us/library/windows/desktop/aa365247%28v=vs.85%29.aspx#naming_conventions): CON, PRN, AUX, NUL, COM1, COM2, COM3, COM4, COM5, COM6, COM7, COM8, COM9, LPT1, LPT2, LPT3, LPT4, LPT5, LPT6, LPT7, LPT8, and LPT9.
* *NIX: `.` and `..` and files [starting with either `-` or `--`](https://stackoverflow.com/a/122427).