arduino/Arduino #6613: Why does PC2 equal 2?

Sod-Almighty, 2017-08-14:
According to the [ATmega328 pinout](https://camo.githubusercontent.com/0a921969a9bbbd74b6d85ffe6cd10f0a5963987a/687474703a2f2f692e696d6775722e636f6d2f624f4a504e35682e6a7067), PC2 is equal to A2 - or to put it another way, Arduino pin 16.

However, according to the Arduino IDE, PC2 is equal to 2:
```
void setup() {
  Serial.begin(115200);
  Serial.println(PC2);
}
```
> 2

This kind of elementary mistake makes it very difficult to get anything done.

per1234, 2017-08-15:
The PC2 macro is defined as 2 because it's bit two of the Port C Data Register.
>PC2 is equal to A2 

Wrong. that pinout is showing that the pin happens to be arbitrarily mapped to Arduino pin number A2/16. There is no relation to the value of the PC2 macro. Arduino functions take Arduino pin numbers as arguments so you can use A2 or 16 but you can't use PC2. If you want to do direct port manipulation you're free to do so but don't get that mixed up with the Arduino pins API.

closed 2017-08-15 by per1234

Sod-Almighty, 2017-08-15:
Fair enough, I didn't realise that. Thanks for the heads up.