arduino/Arduino #6614: Arduino 101 / Arduino 101 stuck after uploading code

tamirmha, 2017-08-15:
Hello,

i have an arduino 101 and after i added to my project an EEPROM Get/Put its worked for 1 minute and after this the device stucked and didn't response anymore and the computer doesnt recognize it any more ( i disconnected from the usb, tried the master reset, tried to downgrade for older version the ide and the drivers).
i tried before to write the only to the eeprom and its worked fine
i have win7 64bit and the latest version of the ide and the intel curie drivers)
I attached the code.
Hope if anyone could help... #bug #arduino101

facchinm, 2017-08-16:
Hi @tamirmha ,
you forgot to attach the sketch :slightly_smiling_face: 
I can help debugging your issue but it would be better to move the discussion to the proper repo (https://github.com/01org/corelibs-arduino101). Would you mind closing this issue and reopening there? Thanks!

tamirmha, 2017-08-16:
Thanks, i posted a new one.

closed 2017-08-16 by tamirmha

per1234, 2017-09-11:
Just to link things together, here's the relevant issue in the other repository: https://github.com/01org/corelibs-arduino101/issues/565